bigtreetech/BIGTREETECH-TouchScreenFirmware #661: [BUG] Entering move menu causes immediate movement

Joeskaife, 2020-05-07:
### Description

Immediately on entering the Move menu, one stepper (usually X) immediately moves a random amount, usually a large positive amount.

### Steps to reproduce

Enter Move menu by touching the Move button and Then Move again.

**Expected behavior**
Nothing - until specific movement requested

**Actual behavior**
A stepper moves.

### Hardware Variant
TFT35_E3_V3
### TFT Firmware Version & Main Board Firmware details
TFT26.1 Unified build, Marlin 2.0.5.3

### Additional Information
If I comment out line 170 in Move.c (ie the storeMoveCmd() call on encoder input) I do not get this movement so I guess it's related to some changes in the encoder handling.

Joeskaife, 2020-05-07:
If I add a call to loopCheckEncoder() in the place where the fix for #560 removed a call to LCD_LoopEncoder() then the movement doesn't happen - though neither does the encoder cause movement so it would be an imperfect fix.

oldman4U, 2020-06-15:
Hello Joe.

Could you please let us know the status of the reported issue. It should be fixed in the meantime.

Thank you

oldman4U, 2020-09-03:
Hi.

Please close this ticket now.

Joeskaife, 2020-09-03:
No idea if this is now fixed, I now use my own patched build for stability, but closing issue on request.

closed 2020-09-03 by Joeskaife