bigtreetech/BIGTREETECH-TouchScreenFirmware #665: [BUG] M109 is not handled correctly

creamywhite, 2020-05-09:
M109 (gcode that instructs the machine to wait for hotend target temperature to be reached) is ignored if printing from SD/USB on TFT. It is skipped which results in multiple "cold extrusion prevented" warnings and machine just does goes on with printing moves without actually extruding anything.

rebel1, 2020-05-09:
Yes, ican confirm this bug after commit https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/522911e738348318132f9c67ac915d93d0dd542e

creamywhite, 2020-05-09:
@rebel1 , thank you for pointing that out. I was able to pinpoint the source of the issue. Changes to TFT/src/User/Menu/Heat.c were done quite sloppily there. I fixed it (verified on my own TFT35 V3.0 E3) in ce4091deb2b78415364c44780515ff364194e5fd and created a pull request https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/670

roomme13, 2020-05-10:
Did #670 fix the issue? i face same issue here.

GregSKR, 2020-05-10:
Yes, it did for me, TFT35 V2.0

oldman4U, 2020-06-15:
Hello creamywhite.

Could you please close all your solved tickets including this.

Thank you

closed 2020-06-15 by creamywhite