bigtreetech/BIGTREETECH-TouchScreenFirmware #668: I found two bugs in the new firmware

AntoszHUN, 2020-05-09:
One major error is turning on the ST7920_SHOW_BANNER option, typing ST7920_BANNER_TEXT instead of displaying "AnRap 8 alpha", PRINT_START_GCODE is displayed showing G28 XY R10. So something was really spoiled here.
[image]

The next mistake is a mystery to me. In the menu shown in the picture, the option for 2 options is incomprehensible. LCD brightness and LCD dimming are incomprehensible.
[image]
[image]
[image]
[image]

Thanks for the reply and correction as soon as possible!

Talha909, 2020-05-09:
i have the same issue

AntoszHUN, 2020-05-09:
Are these mistakes for you too?

Mistigris18, 2020-05-09:
Moi aussi 😞

guruathwal, 2020-05-09:
Similar to issue #664 . Try recompiling and flashing the firmware and reloading the config file.

AntoszHUN, 2020-05-09:
I'm already over it. Nothing has changed. I'd rather put the old firmware back.

guruathwal, 2020-05-09:
did you flashed the icons and fonts also?? the address of icons and fonts was changed after this firmware update.

guruathwal, 2020-05-09:
I have encountered the preheat name error. after re-uploading the config file it was fixed.

AntoszHUN, 2020-05-09:
I updated everything. All files are in place and up to date. If I don't update and only translate the downloaded version, that's the problem.

guruathwal, 2020-05-09:
ok .. let me check

AntoszHUN, 2020-05-09:
All right.

guruathwal, 2020-05-09:
I checked but I  was unable to replicate the ST7920_BANNER_TEXT error. 
you should re-load the config file. you should see update progress if the config file is detected by the TFT.

AntoszHUN, 2020-05-09:
I just don't understand. I downloaded the firmware. I didn't change anything in it. I translated it to TFT35 V3.0 display. I installed. This is a mistake. Could Marlin 2.0.5.3 bother you?

iMike78, 2020-05-09:
We agreed that on fb, couldn't be the Marlin, as I have the same version, but the TFT35_E3_v3, and everything is working just perfect here.

guruathwal, 2020-05-09:
Because the new TFT firmware detects onboard SDcard if it receives `Cap:SDCARD:1` from Marlin firmware.
which is currently only available in the Marlin bugfix branch. it is yet to be merged in Marlin Release branch

guruathwal, 2020-05-09:
@AntoszHUN  can you try re-uploading the config file? 
TFT35_V3 and TFT35_E3_v3 are both the same. the TFT35_E3_v3 only have an extra knob light.

iMike78, 2020-05-09:
Here are the pics to prove that the TFT35_E3_v3 is working fine with Marlin 2.0.5.3
[image]
[image]
[image]
[image]

AntoszHUN, 2020-05-09:
Okay. I don't have time for that. Then next time if the bug is fixed I will use it. Obviously, I’m not the unlucky one if more people have this mistake.

rebel1, 2020-05-09:
> One major error is turning on the ST7920_SHOW_BANNER option, typing ST7920_BANNER_TEXT instead of displaying "AnRap 8 alpha", PRINT_START_GCODE is displayed showing G28 XY R10. So something was really spoiled here.
> [image]
> 
> The next mistake is a mystery to me. In the menu shown in the picture, the option for 2 options is incomprehensible. LCD brightness and LCD dimming are incomprehensible.
> [image]
> [image]
> [image]
> [image]
> 
> I discovered another bug. That's very annoying.
> [image]
> 
> Thanks for the reply and correction as soon as possible!

I have absolutely the same bugs on my TFT35_V3.0 as you. I reflashed everthing and loaded a config.ini on the display, but these two bugs does never change here.

AntoszHUN, 2020-05-09:
I hope it will be remedied soon. I've tried everything. No changes.

rightawy, 2020-05-10:
Reloading the config.ini on TFT fixes it.

guruathwal, 2020-05-10:
Last night I pushed some changes in PR #663 . Can you check if it helps?

AntoszHUN, 2020-05-10:
How can I download it?

SonBroku, 2020-05-10:
> 
> 
> Last night I pushed some changes in PR #663 . Can you check if it helps?

I installed your branch, and there's still the bug where the lcd dim settings are unreadable. Other than that, I can't find any other issues. The bug with the LCD simulator mode banner seems to be fixed. I am getting the firmware yelling at me about no bed leveling data stored, but I think I just need to run a bed level routine.

guruathwal, 2020-05-10:
@SonBroku 
>and there's still the bug where the lcd dim settings are unreadable.

can you give more detail?

SonBroku, 2020-05-10:
> 
> 
> @SonBroku
> 
> > and there's still the bug where the lcd dim settings are unreadable.
> 
> can you give more detail?

Yeah, in the "Feature" settings menu, the settings "LCD brightness," and "LCD Brightness Dim," are both showing nonsense characters. They start out as a 4 with a box next to them, and once changed they just become an empty character with no way to know that the setting is currently set to without counting the amount you click them. In the OP message he has a picture of exactly what I'm talking about. It does this no matter how many times I re-flash or reload the config.ini and I can't seem to find a configuration where this doesn't happen.  I am using the most current marlin bugfix branch. I can't speak to if this is happening on the stable branch, although this seems like a bug with the touchscreen firmware and not marlin.

guruathwal, 2020-05-10:
@SonBroku try reset settings

AntoszHUN, 2020-05-10:
I realized it had nothing to do with Marlin. I removed the connectors from the motherboard. I reinstalled the firmware on the display. I just gave him electricity. Even so, this is a mistake.

@SonBroku
It doesn't belong here.
What is your version of Marlin? Because I am using version 2.0.5.3. But can bugfix 2.0.x be fresher?

AntoszHUN, 2020-05-10:
@guruathwal 
I've already restored everything. It wasn't good either.

guruathwal, 2020-05-10:
@AntoszHUN  did you tried the #663 ?
@SonBroku  reported that it fixed the issue you are facing.

SonBroku, 2020-05-10:
@guruathwal 
Resetting the settings does nothing, It still shows the same gibbirish. This bug doesn't occur with the pre-compiled 26.1 binary.

@AntoszHUN 
I'm using the most current version of the bugfix branch. I wouldn't recommend using the bugfix branch unless you're ok with having potential issues with marlin itself. I'm almost 100% sure this weird menu bug has nothing to do with marlin though. Try downloading and compiling his branch to fix the weird issue with the LCD simulator though, it DOES fix that bug.

AntoszHUN, 2020-05-10:
@guruathwal How do I try it? Not in the firmware.

AntoszHUN, 2020-05-10:
Oh, of course, I can copy it. I'm stupid.

AntoszHUN, 2020-05-10:
@SonBroku I understand. Thanks. Then I drop version 2.0.5.3. I'm reverting to bugfix 2.0.x.

SonBroku, 2020-05-10:
@AntoszHUN Clone his branch of the tft firmware here and compile/flash it:
 https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware/tree/Mainstream

That will fix the bug with the LCD simulator banner text, at the very least. The LCD Dim settings issue seems to be still there. It has yet to be merged into the master branch, but it DOES fix that bug.

AntoszHUN, 2020-05-10:
@guruathwal An error was resolved thanks to # 663.
[image]

De a fényerő és a halványítás még mindig nem jó.
[image]
[image]

guruathwal, 2020-05-10:
Pushed another commit to resolve the issue. please check and revert.

SonBroku, 2020-05-10:
@guruathwal Yep, that fixed it. Cool

AntoszHUN, 2020-05-10:
Thank you Master! This error has also been fixed! Thanks! Thanks! Very good!

Can I close the topic?
[image]

PR #663 (2020-05-23): Add detection of Mainboard firmware type( Marlin or Smoothieware), auto-detect LONG_FILENAME_HOST_SUPPORT, on-boardSD and serial_always_on option in config file & other bugfixes
- Add detection of Mainboard firmware type( Marlin or Smoothieware)
Smoothieware does not support M115 Detailed reporting. Hence some settings need to be enabled or disabled in TFT for Smoothieware.
- Fix text display errors.
- Fix to prevent settings reset after config update.
- add Onboard SD card options for non-marlin firmware in the config file.
- add serial always On option in the config file.
- add missing Hoten [... cut]
closed 2020-05-23 by bigtreetech