bigtreetech/BIGTREETECH-TouchScreenFirmware #669: no onboardsd

Mellux, 2020-05-09:
**Your question or details you need. Please describe.**
A clear and concise question or detail you would like to ask. Like Hardware related or firmware related.

**specify the device variant related to your question.**

**Additional context**
Add any other context or screenshots about the question here.
hi, because when I compile the fw of my tft35 e3 v3 I don't see the icon of the sd card onboard, what do I have to activate in the firmware to have
[image]

guruathwal, 2020-05-09:
you need to enable SD card in Marlin firmware

Mellux, 2020-05-09:
on the firmware of the mainboard is enabled, if I put the fw stock of the display the sd onboard appears but if I compile it I am not there

AntoszHUN, 2020-05-09:
It's on with me. Still, this is a mistake. I don't see the motherboard SD card either.

guruathwal, 2020-05-09:
you also need to update marlin firmware to latest bugfix 2.0x

AntoszHUN, 2020-05-09:
It's not up to me. I am using version 2.0.5.3 Marlin.

guruathwal, 2020-05-09:
SDCard reporting in the Marlin firmware is currently not yet merged in the release branch https://github.com/MarlinFirmware/Marlin/pull/17208
You will have to use the Marlin bugfix branch [here](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) until it is merged or you can use the Release branch of TFT firmware [here](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/release-xx.26.1).

AntoszHUN, 2020-05-09:
Is it better to use bugfix 2.0.x? Don't use version 2.0.5.3?

guruathwal, 2020-05-09:
The new TFT firmware detects onboard SDcard if it receives Cap:SDCARD:1 from Marlin firmware.
which is currently only available in the Marlin bugfix branch. it is yet to be merged in Marlin Release branch.

It is up to you to decide which is better.

AntoszHUN, 2020-05-09:
I understand. Obviously, based on version number, one thinks version 2.0.5.3 is fresher. But it looks like bugfix 2.0.x is much fresher. Then we'll get back to that. I don't even understand why you have to put up the semi-finished version. I need a fix pack and a bugfix. I would only skip the version number if I was sure the firmware was good. So the version numbers are now a bit misleading.

Mellux, 2020-05-09:
thank you very much, I put the bugfix version and the sdonboard is back

AntoszHUN, 2020-05-09:
This proves that 2.0.5.3 is not newer than bugfix 2.0.x. So the bigger version number is meaningless! Pls closed. Solved for me too!

guruathwal, 2020-05-09:
@AntoszHUN 
To be clear you are currently using the bugfix/dev branch of the TFT firmware, it is just named `master` as BTT has not renamed it. 
bugfix branches are always unfinished.
If you want a finished version, then use the release branch.

@Msq001 I recommend renaming the `master` branch to `dev` or `bugfix` branch to remove the confusion.

AntoszHUN, 2020-05-09:
I understand.

cwizard, 2020-05-09:
Can Confirm using Marlin Bugfix branch with TFT "Master" branch gives me print from onboard SD functionality again.

oldman4U, 2020-05-09:
Hi.

For me the versioning system is unclear too, but you can see for every version how many commits each version is behind or ahead of another release and you can see which functions the commits are related to.

This means more for me than a release number and allows to download a specific release.

[image]
[image]
[image]

AntoszHUN, 2020-05-10:
I always update the Master branch. The Hungarian language and what I do. Everything goes to the Master branch.

oldman4U, 2020-06-15:
Hello Mellux.

Could you please close this ticket.

Thank you

closed 2020-07-04 by bigtreetech