bigtreetech/BIGTREETECH-TouchScreenFirmware #671: [FR] Separation between banner and 12864 info

zillarob, 2020-05-09:
The banner sits right on top of the 12864 display with the adjacent pixels touching.

Kinda looked funny so added this to give them a little separation.
I dont know enough about this stuff to know if this is a good or bad way of going about it, but seems to work.

[image]

zillarob, 2020-05-11:
Found the one for y start on the 12864 part and like it better.
Adding +8 to it gets it down to the bottom more and away from the banner.

A conditional might be nice, like
ifdef ST7920_BANNER_TEXT
+8 to shift it down toward the bottom more
else
+0 to have the screen centered like normal

[image]

zillarob, 2020-05-11:
Here is a pic. Normally the banner would be sitting right on top of the temp, and touching it.

[image]

oldman4U, 2020-10-09:
Seems like this problem is solved. 

Please help the community and close the ticket once you do not need it anymore. Thank you

oldman4U, 2020-10-20:
@zillarob 

Please close this ticket.

closed 2020-12-26 by stale[bot]