bigtreetech/BIGTREETECH-TouchScreenFirmware #675: [BUG] Preheat buttons 2 & 3 (PETG and ABS by default) are not working

creamywhite, 2020-05-10:
Preheat buttons (Menu - Heat/Fan - Preheat) number 2 and 3 (PETG and ABS by default) are not working. They do not set any heating targets for hotend/bed/both after clicking them. As a note, first one (PLA) works as intended for hotend/bed/both.

SonBroku, 2020-05-11:
Confirm, I'm also having this issue. I'm gonna take a looksee and see what I can find, if there's anything obvious. I'll let you know if I come up with anything.

SonBroku, 2020-05-11:
Okay, it's actually working, but it's a bit finicky. When I tap so that the bottom icon says "both" and tap PETG, it will start heating both the nozzle and bed, but not until I back out of that menu. 

Then when I go back into the menu, it behaves totally differently. It seems like the entire implementation of that menu is kind of weird and buggy.  I would say you're correct in that the PLA button mostly behaves as normal, but the PETG and ABS ones DO work, they're just...kind of weird.

creamywhite, 2020-05-11:
@SonBroku , I'm unable to make them work like you've described.

roomme13, 2020-05-11:
in my build, that button work fine, but the label is screw-up.
[image]

creamywhite, 2020-05-11:
@roomme13 , had that problem too (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/664) - somehow, rebuilding the FW and reloading the config helped. However, I don't know if all preheat buttons worked for me when their labels were mangled like that - didn't test it then.

roomme13, 2020-05-11:
I just build this one recently, 2 or 3 hours ago. Same result, maybe I should try it one more time.
In my case that just label, the function itself work  normally.
Thank you by the way.

closed 2020-08-19 by bigtreetech