bigtreetech/BIGTREETECH-TouchScreenFirmware #676: [Q] (Unable to install on MKS TFT28 v3.0)

Plizze, 2020-05-10:
I have copied the file "BIGTREE_TFT28_V3.0.26.1.bin" + "TFT28" folder to the SD but it does not work.
- I have also tried with "BIQU_TFT28_V1.0.26.1.bin", file, but the same...
- Copying to SD: "BIGTREE_TFT28_V3.0.26.1.bin" + "BIQU_TFT28_V1.0.26.1.bin"... nothing
Extracting"bmp" and "font" folders from TFT28 folder to SD root and enaming these as "mks_font" and "mks_pic".nothing...
The TFT28 screen always stops showing "Updating 100%" but does not advance.

oldman4U, 2020-05-10:
Hello Plizze.

As far as i understand, the following screens are currently supported by the BTT firmware:

oldman4U, 2020-05-10:
;BIGTREE_TFT35_V1_0
;BIGTREE_TFT35_V1_1
;BIGTREE_TFT35_V1_2
;BIGTREE_TFT35_V2_0
;BIGTREE_TFT35_V3_0
;BIGTREE_TFT35_E3_V3_0
;BIGTREE_TFT28_V1_0
;BIGTREE_TFT28_V3_0
;BIGTREE_TFT24_V1_1
;MKS_32_V1_4
;MKS_32_V1_4_NOBL

Hope this helps

thisiskeithb, 2020-05-12:
The MKS TFT28 is not supported in this firmware. You’ll need to use the firmware from [MKS](https://github.com/makerbase-mks/MKS-TFT).

Uz45, 2020-05-12:
> I have copied the file "BIGTREE_TFT28_V3.0.26.1.bin" + "TFT28" folder to the SD but it does not work.
>
> * I have also tried with "BIQU_TFT28_V1.0.26.1.bin", file, but the same...
> * Copying to SD: [...]

For MKS TFT28 3.0 use [this fork ](https://github.com/Uz45/BIGTREETECH-TouchScreenFirmware/tree/BTT_261_MKSTFT32_V1_V2_V3_V4) MKS TFT32 V
MKS TFT32 and MKS TFT28 use the same firmware.

Plizze, 2020-05-14:
Thank you all for the information.

digant73, 2020-05-27:
> > I have copied the file "BIGTREE_TFT28_V3.0.26.1.bin" + "TFT28" folder to the SD but it does not work.
> > 
> > * I have also tried with "BIQU_TFT28_V1.0.26.1.bin", file, but the same...
> > * Copying to SD: "BIGTREE_TFT28_V3.0.26.1.bin" + "BIQU_TFT28_V1.0.26.1.bin"... nothing
> >   Extracting"bmp" and "font" folders from TFT28 folder to SD root and enaming these as "mks_font" and "mks_pic".nothing...
> >   The TFT28 screen always stops showing "Updating 100%" but does not advance.
> 
> For MKS TFT28 3.0 use [this fork ](https://github.com/Uz45/BIGTREETECH-TouchScreenFirmware/tree/BTT_261_MKSTFT32_V1_V2_V3_V4)
> MKS TFT32 and MKS TFT28 use the same firmware.

@Uz45 I see from your branch that one the change is on the "spi_slave.c" file and on "pin_MKS_TFT32_V1_4.h" (My Artillery Sidewinder X1 should use MKSTFT28 v1.4 if I'm not wrong) were you configured ST7920_SPI to _SP3. Do these changes mean that Marlin mode is supported also in MKS TFT32 1.4?

Uz45, 2020-05-28:
> @Uz45 I see from your branch that one the change is on the "spi_slave.c" file and on "pin_MKS_TFT32_V1_4.h" (My Artillery Sidewinder X1 should use MKSTFT28 v1.4 if I'm not wrong) were you configured [...]

	To support Marlin mode with MKS TFT screens, a screen modification is required.
	You need to solder one wire to the STM32 89 pin, this is the PB3 pin for SPI3.
	Solder and fix the wire, STM32 pins are very easy to break off.
	I soldered this wire to the NC pin on the WIFI connector,this can be seen in the 
	photo mkstft32_12864.jpg.
	For more information, see the Readme_for_MKS file TFT32.txt

digant73, 2020-05-28:
> > > > I have copied the file "BIGTREE_TFT28_V3.0.26.1.bin" + "TFT28" folder to the SD but it does not work.
> > > > 
> > > > * I have also tried with "BIQU_TFT28_V1.0.26.1.bin", file, but the same...
> > > > * Copying to SD: "BIGTREE_TFT28_V3.0.26.1.bin" + "BIQU_TFT28_V1.0.26.1.bin"... nothing
> > > >   Extracting"bmp" and "font" folders from TFT28 folder to SD root and enaming these as "mks_font" and "mks_pic".nothing...
> > > >   The TFT28 screen always stops showing "Updating 100%" but does not advance.
> > > 
> > > 
> > > For MKS TFT28 3.0 use [this fork ](https://github.com/Uz45/BIGTREETECH-TouchScreenFirmware/tree/BTT_261_MKSTFT32_V1_V2_V3_V4)
> > > MKS TFT32 and MKS TFT28 use the same firmware.
> > 
> > 
> > @Uz45 I see from your branch that one the change is on the "spi_slave.c" file and on "pin_MKS_TFT32_V1_4.h" (My Artillery Sidewinder X1 should use MKSTFT28 v1.4 if I'm not wrong) were you configured ST7920_SPI to _SP3. Do these changes mean that Marlin mode is supported also in MKS TFT32 1.4?
> 
> ```
> To support Marlin mode with MKS TFT screens, a screen modification is required.
> You need to solder one wire to the STM32 89 pin, this is the PB3 pin for SPI3.
> Solder and fix the wire, STM32 pins are very easy to break off.
> I soldered this wire to the NC pin on the WIFI connector,this can be seen in the 
> photo mkstft32_12864.jpg.
> For more information, see the Readme_for_MKS TFT32.txt file
> ```

Thanks for the clarification. So in order to use the emulation I only need to:
1) bridge 2 pins on the PCB (no problem for that)
2) connect some pins on the PCB with dupont cables
3) make some configuration changes on both the TFT FW (on pin_MKS_TFT32_V1_4.h file) and also in Marlin FW
4) noaddition hardware is needed

Right? If so, I will make the changes

Furthermore, as reported on bug #723 I got many errors printing from Pronterface/Octoprint. Pretty no errors if I print from Simplify3D (that should be a built in Pronterface, if I'm not wrong).
I get the problem with both your modified code and the stock BTT code.
Please, can you let me know if you have the same problems?
It seams that in the communication between TFT and Marlin something is lost (maybe the TFT is so much stressed and is not able to reply to Marlin correctly.
As far I see on the configuration file  pin_MKS_TFT32_V1_4.h file, the communication should be on _USART2

Thanks

Uz45, 2020-05-28:
@digant73 

  To print from a computer, use Menu =>Settings=>Disonnect. While printing is in progress, the TFT screen remains inactive and you can't use it to monitor or control the printer.

  No additional hardware is required to operate MKS TFT in 12864 mode. Just output the PB3 signal and connect the screen with a custom cable to your control Board. Unfortunately, I can't provide a photo of the MKS GEN L V1.0 connection, since I only use SKR V1.3 now. But when connecting, I was guided by the following illustration of the pin location.
[image]

digant73, 2020-05-28:
> @digant73
> 
> To print from a computer, use Menu =>Settings=>Disonnect. While printing is in progress, the TFT screen remains inactive and you can't use it to monitor or control the printer.
> 

Yes, I know. I also tested it and it works. So is the problem also present on BTT TFT not just on MKS TFT?

> No additional hardware is required to operate MKS TFT in 12864 mode. Just output the PB3 signal and connect the screen with a custom cable to your control Board. Unfortunately, I can't provide a [...]

Clear. Just a possible change on your Readme_for_MKS TFT32.txt, you reported the ext1 and ext2 connectors pinout on the the example for the SKR 1.3 and not MKS Gen L v1.0. Plus you reported this pinout
               _____                                              _____
            NC | 1 2 | GND                                     5V | 1 2 | GND
         RESET | 3 4 | 1.31 (SD_DETECT)             (LCD_D7) 1.23 | 3 4 | 1.22 (LCD_D6)
   (MOSI) 0.18 | 5 6   3.25 (BTN_EN2)               (LCD_D5) 1.21 | 5 6   1.20 (LCD_D4)
  (SD_SS) 0.16 | 7 8 | 3.26 (BTN_EN1)               (LCD_RS) 1.19 | 7 8 | 1.18 (LCD_EN)
    (SCK) 0.15 | 9 10| 0.17 (MISO)                 (BTN_ENC) 0.28 | 9 10| 1.30 (BEEPER)
                -----                                              -----
                 EXP2                                               EXP1

Compared to the picture you provided before it seems that your connectors must be rotated by 180° to match the pinout in the picture. Am I right?

thanks

Uz45, 2020-05-29:
If you connect MKS TFT to MKS GEN L V1.0, use the following pin diagram.
	MKS TFT32								MKS Gen L V1.0 (Marlin pins)

 LCD_ENCA_PIN  PA13 	JTAG DIO 	=>		exp2 D31-pin (BTN_EN1)
 LCD_ENCB_PIN  PA14 	JTAG CLK	        =>		exp2 D33-pin (BTN_EN2)
 LCD_BTN_PIN   PB0 		PB0		=>		exp1 D35-pin (BTN_ENC)
 SPISIM_CS_PIN PB1 		PB1		=>		exp1 D16-pin (LCD_PINS_RS)
 SPI3_SCK_PIN  PB3 		PB3 		=>		exp1 D23-pin (LCD_PINS_D4)
 SPI3_MOSI_PIN PB5 		PB5		=>		exp1 D17-pin (LCD_PINS_ENABLE)

[image]

 For easy connection, the BTN_EN1 and BTN_EN2 signals must be reassigned to the EXP1 connector. You can do this in Marlin in the pins_RAMPS.h file

About a printing problem from Octoprint.
I think the problem is not on the screen side, the problem is fundamental. When Marlin receives two G-CODES at the same time, which one will be executed? Two drivers driving the same car. I had a problem similar to yours with stock firmware MKS and Pronterface, on different boards. Since printing was done from the screen, it was enough to perform a diskonnect from Pronterface for the solution.

digant73, 2020-05-29:
> If you connect MKS TFT to MKS GEN L V1.0, use the following pin diagram.
> MKS TFT32 => MKS Gen L V1.0 (Marlin pins)
>
> LCD_ENCA_PIN PA13 JTAG DIO => exp2 D31-pin (BTN_EN1)
> LCD_ENCB_PIN PA14 JTAG CLK => [...]

Ok. The pins connection is clear. I'm just preparing the cables.
So the gcode commands sent by pronterface on the USB port of MKS GEN L are received by both Marlin and TFT and TFT will also send the commands to Marlin (so Marlin is receivng twice each gcode command). Is that the use case? Just to better understand the flow

Uz45, 2020-05-29:
> So I could use only ext1 instead of also ext2 (just for 2 pins connection), right? What changes should I make on pins_RAMPS.h? I mean which pin could I use instead of 31 and 33?
> It seem that the remaining pin on ext1 (25, 27, 29 and 37) are already assigned to LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7 and BEEPER_PIN

You can swap pins
#define LCD_PINS_D5 25  <> #define BTN_EN1 31
#define LCD_PINS_D6 27 <> #define BTN_EN2 33

Line numbers may differ depending on the Marlin version. I have them like this

It should work like this

467   #define LCD_PINS_D5 31
468   #define LCD_PINS_D6 33

503  #define BTN_EN1 25
504  #define BTN_EN2 27

> So the gcode commands sent by pronterface on the USB port of MKS GEN L are received by both Marlin and TFT and TFT will also send the commands to Marlin (so Marlin is receivng twice each gcode command). Is that the use case? Just to better understand the flow

I don't know for sure but I assume that the conflict occurs when transmitting from the M105 screen. The screen sends the M105 command every 3 seconds to get a temperature report. Not consistent with the gcode stream from Octoprint, including at the same time with It.

digant73, 2020-05-29:
> You can swap pins
> #define LCD_PINS_D5 25 <> #define BTN_EN1 31
> #define LCD_PINS_D6 27 <> #define BTN_EN2 33
>
> Line numbers may differ depending on the Marlin version. I have them like this
>
> It should [...]
Perfect, I will swap these settings so I can use only exp1 connector.

> > So the gcode commands sent by pronterface on the USB port of MKS GEN L are received by both Marlin and TFT and TFT will also send the commands to Marlin (so Marlin is receivng twice each gcode command). Is that the use case? Just to better understand the flow
> 
> I don't know for sure but I assume that the conflict occurs when transmitting M105 from the screen. The screen sends the M105 command every 3 seconds to get a temperature report. Not consistent with the gcode stream from Octoprint, including at the same time with It.

Yes, I know the TFT sends this kind of gcode. I will try to avoid to send it and see if something changes.

Thanks again

digant73, 2020-05-31:
> BTN_EN1

Hi, I made all the changes on:
1) Marlin FW:
  *) uncommented REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
  *) uncommented and set to 3 ENCODER_PULSES_PER_STEP (by default it was set to 4)
  *) swapped pin 25<->31 and 27 <->33 in order to use only EXP1 connector for the connections with the 6 pins on the TFT
2) TFT FW:
  *) used your pin_MKS_TFT32_V1.4.h file enabling the LCD emulation
3) TFT board:
  *) I wired the pin 89 (PB3) of the main chip on a free pin and connected the 6 pin on the EXP1 connector of the MKS GEN L. Checked with multimeter everything was good (e.g. continuity between pin 89 on the chip with the wired cable etc...)

However, when I start the TFT in Marlin mode I see the title "LCD12864 Simulator" on top of the screen but on the rest of the screen I see most of the time a black screen and sometime wrong chararaters here and there.
Do you have any idea of the problem or what I can check or change in the configuration (Marlin, TFT)?

PS: My Artillery Sidewinder has a clone of the MKS TFT28 v1.4. I hope this is not the cause of the problem

Uz45, 2020-06-01:
@digant73
> Hi, I made all the changes on:
>
> 1. Marlin FW:
>    *) uncommented REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
>    *) uncommented and set to 3 ENCODER_PULSES_PER_STEP (by default it was set to 4) [...]

We should try in the spi_slave.c , replace the string 38 
if defined(MKS_32_V1_4) | | defined(MKS_32_V1_3) | / defined(MKS_32_V1_2) || defined(MKS_32_V1_1)

 with the string 
if defined(MKS_32_V1_3) || defined(MKS_32_V1_2) || defined(MKS_32_V1_1)

Uz45, 2020-06-02:
In the marlin, after # define reprap_discount_smart_controller configuration.h the following defines

#define ST7920_DELAY_1 DELAY_NS(0)

#define ST7920_DELAY_2 DELAY_NS(125)

#define ST7920_DELAY_3 DELAY_NS(0)

digant73, 2020-06-03:
> @digant73
> This solution may help.
> In the marlin, after # define reprap_discount_smart_controller configuration.h the following defines
> 
> #define ST7920_DELAY_1 DELAY_NS(0)
> 
> #define ST7920_DELAY_2 DELAY_NS(125)
> 
> #define ST7920_DELAY_3 DELAY_NS(0)
> 
> You can control the cursor by swiping up and down. To the right of the menu items. Only this area of the screen responds to gestures.

Tried, but same results.
Just to avoid misunderstanding from my side, the setting I changed on configuration.h are:
1) uncommented REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER (not REPRAP_DISCOUNT_SMART_CONTROLLER)
2) uncommented and set ENCODER_PULSES_PER_STEP 3
3) #define ST7920_DELAY_1 DELAY_NS(0)
   #define ST7920_DELAY_2 DELAY_NS(125)
   #define ST7920_DELAY_3 DELAY_NS(0)

Uz45, 2020-06-04:
@digant73
>I just found that if I insert an SD card in the TFT slot, the Marlin menu becomes black (with wrong charaters here and there) even in case 1) described above. Is it possible that there are conflicts with other resources (SPI etc...)?

The SD card uses SPI 1 and there should be no conflicts.
This behavior indicates a possible problem with screen power. The SD card consumes additional power. And the insufficient voltage becomes even less. 

> 1. uncommented REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER (not REPRAP_DISCOUNT_SMART_CONTROLLER)

Yes, that's right, I was in a hurry and made a typo!

It looks like the reason is hardware not software. The cable may be too long. I used the minimum possible length of cable, made it from FDD cable. But a possible power problem is more likely.

digant73, 2020-06-05:
> @digant73
> 
> > I just found that if I insert an SD card in the TFT slot, the Marlin menu becomes black (with wrong charaters here and there) even in case 1) described above. Is it possible that there are conflicts with other resources (SPI etc...)?
> 
> The SD card uses SPI 1 and there should be no conflicts.
> This behavior indicates a possible problem with screen power. The SD card consumes additional power. And the insufficient voltage becomes even less.
> 
> > 1. uncommented REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER (not REPRAP_DISCOUNT_SMART_CONTROLLER)
> 
> Yes, that's right, I was in a hurry and made a typo!
> 
> It looks like the reason is hardware not software. The cable may be too long. I used the minimum possible length of cable, made it from FDD cable. But a possible power problem is more likely.
> 
> You can still try this option.
> #define ST7920_DELAY_1 DELAY_NS(250)
> #define ST7920_DELAY_2 DELAY_NS(250)
> #define ST7920_DELAY_3 DELAY_NS(250)
> 
> Important note! This mod only works as a display, the SD card on the screen will not be available in Marlin.

Yes, I also suspected that the wrong characters displayed were due to defective cables (or some interference) so I changed the 6 dupont cables with an FDD cable as short as possible to connect the TFT to the main board. Now there is no more wrong characters displayed but I have always the same behaviour. I also tried to used very short cables (10cm long) detaching the TFT from the chassis and connecting it very close to the main board. Same result as the FDD cable.
With the new FDD cable, the problem with the SD is not present (even if a put a SD card, no wrong characters are displayed).
Hower in the last chat I forgot to mention that in touch mode I continuosly see the message "U disk inserted" followed by "U disk removed". This problem is stil present even with the new FDD cable.

below you can see some pictures of the new cabling:

[image]
[image]
[image]

Here a video of the Marlin menu that is displayed only when powering the printer by USB:
https://www.youtube.com/watch?v=6NhD8ym5Fy0

As you can see only the OK button on top right corner seems to work. If I try to select a menu on the list of menu nothing happes. Just few time the cursor moved to the second item but I don't know if it was a sporadic (random) case

labotecno, 2020-07-10:
> 
> 
> > @digant73
> > > I just found that if I insert an SD card in the TFT slot, the Marlin menu becomes black (with wrong charaters here and there) even in case 1) described above. Is it possible that there are conflicts with other resources (SPI etc...)?
> > 
> > 
> > The SD card uses SPI 1 and there should be no conflicts.
> > This behavior indicates a possible problem with screen power. The SD card consumes additional power. And the insufficient voltage becomes even less.
> > > 1. uncommented REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER (not REPRAP_DISCOUNT_SMART_CONTROLLER)
> > 
> > 
> > Yes, that's right, I was in a hurry and made a typo!
> > It looks like the reason is hardware not software. The cable may be too long. I used the minimum possible length of cable, made it from FDD cable. But a possible power problem is more likely.
> > You can still try this option.
> > #define ST7920_DELAY_1 DELAY_NS(250)
> > #define ST7920_DELAY_2 DELAY_NS(250)
> > #define ST7920_DELAY_3 DELAY_NS(250)
> > Important note! This mod only works as a display, the SD card on the screen will not be available in Marlin.
> 
> Yes, I also suspected that the wrong characters displayed were due to defective cables (or some interference) so I changed the 6 dupont cables with an FDD cable as short as possible to connect the TFT to the main board. Now there is no more wrong characters displayed and no problem with the SD (even if a put a SD card, no wrong characters are displayed) but I have always the same behaviour (menu only with USB). I also tried to use very short cables (10cm long) detaching the TFT from the chassis and connecting it very close to the main board. Same result as the FDD cable.
> However in the last chat I forgot to mention that in touch mode I continuosly see the message "U disk inserted" followed by "U disk removed". This problem is stil present even with the new FDD cable.
> 
> below you can see some pictures of the new cabling:
> 
> [image]
> [image]
> [image]
> 
> Here a video of the Marlin menu that is displayed only when powering the printer by USB:
> https://www.youtube.com/watch?v=6NhD8ym5Fy0
> 
> As you can see only the OK button on top right corner seems to work. If I try to select a menu from the list of menu nothing happens (unless I'm pushing on the wrong way). Just few time the cursor moved to the second item but I don't know if it was a sporadic (random) case
> 
> Furthermore, no menu will be displayed if I unplug the FDD cable. Does it mean that if some menu is displayed (as it is if the system is powered with USB only) the communication with the Marlin was correctly established through the 6 connections?

did you get working ?

digant73, 2020-07-10:
@labotecno no. I'm waiting for an FDD connector to use in the Marlin board. Do you have the same problem?

labotecno, 2020-07-10:
i have the original mks gen l from artillery , why you need a fdd conector ?

digant73, 2020-07-13:
I will use it on the GEN L board (exp1) instead of using dupont connectors.

labotecno, 2020-07-14:
yes i think you need every pin used for lcd graphic to make it work , its not enough with the connection only for touch

closed 2020-11-03 by stale[bot]