bigtreetech/BIGTREETECH-TouchScreenFirmware #680: When encoder used in touch mode to adjust values, feed rate changes in LCD mode.

SonBroku, 2020-05-11:
As the title says. I have noticed that If I use the encoder knob to adjust values such as the fan speed in menus in the touch screen mode, it's also cranking the feed rate up and down in the LCD mode. So say I crank my fan up from 0 to 100% in the touch menu, if I then go back to LCD mode, the FR will show as something like 175%. It seems like the encoder is being passed to marlin regardless of which mode the screen is in.

guruathwal, 2020-05-11:
This is because the encoder wheel is hard wired to the EXP ports and the mainboard is also reading the encoder status all the time.
With those clicking and rotating of the encoder wheel in touch mode, the mainboard might end up in serious damage to the machine, like heating being enabled, increase temperature, or accidentally moving any axis.

creamywhite, 2020-05-11:
Then it seems like a major hardware design oversight to me. I guess MOSFETs should've been used on encoder's VCC or GND lines to disable its operation when the TFT is used in Touch mode.

SonBroku, 2020-05-12:
Okay so using the encoder to adjust values in touch mode is a no go because it's got a direct connection to the printer through the EXP connection. That does seem like a bit of an oversight, but using the encoder to change values in touch mode isn't exactly necessary, I'm just used to it coming from the stock ender 3 LCD. I find myself using LCD simulation more than the touch mode because I'm so used to It.

SonBroku, 2020-05-12:
I guess I'll close the issue as it seems to be an inherent hardware limitation of the board. Oh well.

closed 2020-05-12 by SonBroku

radek8, 2020-05-13:
Hi
I found a solution to this problem
add a line wherever the encoder is used and where I do not want it to change the feed rate

sendEncoder (2);

[image]

PR #1203 (2020-10-27): Change ENC_ACTIVE_SIGNAL logic to disable encoder in Marlin firmware when TFT is not in Marlin mode.
- Change `ENC_ACTIVE_SIGNAL` logic to disable encoder in Marlin firmware when TFT is not in Marlin mode. Details in https://github.com/MarlinFirmware/Marlin/pull/19796
- clean-up
- Add stale bot config to manage old abandoned issues. @bigtreetech Needs installing stale app for the repository from here: https://github.com/apps/stale

This will fix #680 after https://github.com/MarlinFirmware/Marlin/pull/19890 is merged in marlin firmware.
Resolves #1206 , Resolves #1045