bigtreetech/BIGTREETECH-TouchScreenFirmware #681: [FR] IDEA: Iconpack 'The Round Miracle' - Update for SW v26.1

Acenotass, 2020-05-11:
Updated, already liked by you the iconpack 'The Round Miracle'!

[image]

[TFT24 Iconpack 'The Round Miracle' by Acenotass for v26.1.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4608152/TFT24.Iconpack.The.Round.Miracle.by.Acenotass.for.v26.1.zip)
[TFT28 Iconpack 'The Round Miracle' by Acenotass for v26.1.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4608153/TFT28.Iconpack.The.Round.Miracle.by.Acenotass.for.v26.1.zip)
[TFT35 Iconpack 'The Round Miracle' by Acenotass for v26.1.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4608154/TFT35.Iconpack.The.Round.Miracle.by.Acenotass.for.v26.1.zip)

Kyosuke33, 2020-05-22:
Hello, i have an error with the TFT24 pack. It tell me than files are missing, and on the screen, the 3 first icone are missing.

Kyosuke33, 2020-05-24:
[image]
[image]

Kyosuke33, 2020-05-24:
Statut bed too butni miss the picture.

Kyosuke33, 2020-05-24:
Edit : i try with original firmware, same problem, so this is not about your pack. Thank you.

Kyosuke33, 2020-05-24:
Edit : It's ok, it's because on the firmware they made a mistake on the name.

Status_Fan -> StatusFan
 Status_Bed -> StatusBed
 Status_Nozzle -> StatusNozzle

oldman4U, 2020-06-04:
Hello acenotass.

I am using your theme on my printers and I really like what you did. It seems that there will be an extra bmp needed very soon, which shows the temp of the Nozzle and the Bed at the same time. Maybe you already know this - if not, attached is the standard bmp.

Thank you
[preheat.bmp.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4732456/preheat.bmp.zip)

oldman4U, 2020-07-09:
Hi acenotass. 

The new software is out. It would be great if you could provide the missing preheat.bmp. Thank you

Acenotass, 2020-07-10:
@oldman4U https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/832

Acenotass, 2020-07-10:
New version

closed 2020-07-10 by Acenotass