bigtreetech/BIGTREETECH-TouchScreenFirmware #682: [BUG] BMP naming differences 25.4 and 26.1

cht47, 2020-05-11:
### Description

If you flash 0.25.4 you get errors that some bmp could not be found. 

StatusFan -> Status_Fan
StatusBed -> Status_Bed
StatusNozzle -> Status_Nozzle

### Steps to reproduce

1. flash old firmware with new pictures

### Solution
either split the image resources so that the old firmware continues to work without problems or restore the old names in the new firmware and rename the three files.

But the best solution would be to TAG 0.25.4 the last commit before renaming and remove the bin from the master. 

(you have to use git correctly otherwise you can move it to dropbox)

SonBroku, 2020-05-12:
They really need to clean up this repo. It's a mess to be honest. There's 2 releases with incompatibilities in the precompiled binaries. They should just remove the 0.25.4 ones or put them in their own old branch. I'm having to use my own fork to amalgamate fixes I'm reading in these issues threads that aren't even being submitted with pull requests.

guruathwal, 2020-05-12:
@cht47  You can get release versions (including source code and all icons) from **Releases** page: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/releases

@SonBroku The files you are mentioning are not Release files. Releases are in **Releases** page mentioned above. You are looking at the development branch where the code needs to be compiled by the user to get the correct firmware binary. 
Bugs are expected as code is being added and removed by many programmers who are not working in sync. There will be times when everything will be messed up and times when everything works as expected. 
To avoid the mess you can get a stable version from the **Release** page.

SonBroku, 2020-05-13:
@guruathwal I'm aware of the release branch. I encountered serious bugs with it, so I had to switch to using the master branch with various fixes. It doesn't help that if you google "bigtreetech tft35 e3 firmware" the first result is the master branch on github. I understand it's all a process, I'm just frustrated at how difficult it has been to get this thing running smoothly, but it's getting better with time.

oldman4U, 2020-06-15:
Hello cht47.

Have you been able to upload the bmp's correctly in the meantime?

Please let us know

Thank you

cht47, 2020-06-17:
> Hello cht47.
> 
> Have you been able to upload the bmp's correctly in the meantime?
> 
> Please let us know
> 
> Thank you

I'm using a newer build and I flashed it with the correct names. I'm using now my own "picture repo" with some color changes and name fix.

oldman4U, 2020-06-17:
Sounds great.

Could you please close this ticket, seems you do not need it anymore.

Thank you

closed 2020-08-19 by bigtreetech