bigtreetech/BIGTREETECH-TouchScreenFirmware #685: [BUG] Marlin Screen & EXP3 cable prevents printer startup (TFT35 E3 V3.0 w/ SKR 1.3)

NickTom88, 2020-05-11:
### Description

After installing my new TFT35 E3 V3.0, On my Ender 3 Pro, running Marling 2.0.5.3, I updated to the most recent firmware, hooked up both cables and turned on my printer. At first everything seemed fine with the Classic Touch Screen. Once I switched to Marlin, I noticed nothing would pop up or work. This was caused by the EXP cable not being in the correct port. Once I placed it in EXP3, everything came on and worked correctly. 

After I turned the printer off and the back on, nothing seemed to be starting up. I unplugged the EXP3 cable and suddenly the printer booted up. I plugged the EXP3 cable back in and everything was operating correctly.

It appears that something with the EXP port is preventing the printer from booting up...

### Steps to reproduce

1. One Ender 3 Pro or printer using Marlin 2.0.5.3 Flash the firmware to BIGTREE_TFT35_V3.0_E3.26.1.bin with Classic Touch Screen
2. Plug in R232 and EXP3 cables
3. Power Cycle printer
4.

**Expected behavior**
Printer boots up and can cycle through CTS and Marlin menus

**Actual behavior**
The printer will not boot up like normal and the screen shouldn't turn on until you unplug the EXP3 cable.

### Hardware Variant
TFT35 E3 V3.0

### TFT Firmware Version & Main Board Firmware details
TFT: BIGTREE_TFT35_V3.0_E3.26.1
Main Board: SKR 1.3 w/ marlin 2.0.5.3

### Additional Information
- Cannot currently supply config files since I'm at work. Will upload ASAP.

* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

gilliands, 2020-05-14:
Did you remember to not connect the Reset line on the TFT serial link? Having a reset signal on EXP3 and another on the TFT serial results in a reset loop.  I left the TFT  Reset signal unconnected and all worked as normal.  (Using SKR 1.3 + TFT 35 E3 + ESP01S(16mbit))

NickTom88, 2020-05-23:
Sorry for the late reply here. Been distracted with other things. 

But you were correct. Out of all the research I did this never came up.

biga888, 2020-05-27:
Thank you, I did not find anything else about this either. Classic mode works now

oldman4U, 2020-06-08:
Hi @NickTom88.

Seems the problem could be solved. Could you please close this ticket in case you do not need it anymore.

Thank you

NickTom88, 2020-06-08:
> Hi @NickTom88.
> 
> Seems the problem could be solved. Could you please close this ticket in case you do not need it anymore.
> 
> Thank you

@oldman4U weird. I swear I closed this  at least a week ago.

closed 2020-06-08 by NickTom88

oldman4U, 2020-06-08:
I will check again tomorrow 😁