bigtreetech/BIGTREETECH-TouchScreenFirmware #686: [BUG] Probe Z offset adjustment in touch mode does not work.

SonBroku, 2020-05-12:
### Description

### Steps to reproduce

1. Try to use the Z probe offset adjustment in the touch screen menu.

**Expected behavior**

The nozzle moves up and down so you can use the paper method to properly set your probe Z offset.
**Actual behavior**

The nozzle doesn't move at all, it seems as if it's not even changing the offset.
### Hardware Variant

### TFT Firmware Version & Main Board Firmware details
TFT35 e3 V3.0

Marlin 2.0.x bugfix 
current compiled TFT35 firmware from here on github.
### Additional Information

SonBroku, 2020-05-12:
Ok, I noticed that if I use the "babystep" menu, it will adjust the probe offset as well. I'm guessing this has to do with the option to combine babystepping and z probe offset in marlin.

SonBroku, 2020-05-12:
With `//#define BABYSTEP_ZPROBE_OFFSET`, behaviour is the same. I can set the offset by using the "babystep" menu, but the actual probe offset doesn't seem to move the nozzle in real time like the one in marlin would.

guruathwal, 2020-05-12:
the Z Offset menu is for setting the nozzle z offset values from the probe tip by using **M851** G-code.
>Expected behavior
>
>The nozzle moves up and down so you can use the paper method to properly set your probe Z offset.

why do you expect **M851** should move the nozzle?

for moving the Nozzle there is already a Move menu in TFT.

SonBroku, 2020-05-12:
Because that's how it works in marlin. The probe Z offset menu moves the nozzle as you change it. When adjusting the first layer, you can move it up and down just like the babystep menu to adjust the nozzle to bed distance.

SonBroku, 2020-05-12:
I guess I’ll just close the issue. It seems like an inherent problem in that the touch screen mode is essentially just a print host and as such can’t take advantage of some of the weird marlin UI options. Upon further research it seems that I’m just used to how marlin combines the babystep menu with the z probe offset menu. Behind the scenes it seems marlin is doing this by mapping offset adjustments to babystep moves. Oh well. I’ll get used to just using the babystep menu.

digant73, 2020-05-13:
> the Z Offset menu is for setting the nozzle z offset values from the probe tip by using **M851** G-code.
>
>
> why do you expect **M851** should move the nozzle?
>
> for moving the Nozzle there is already [...]

however when using babystep the z-offset is updated to value 0.0 due to the bug reported in #626. There is a missing ":" charater when parsing the echo message. That bug was not yet fixed

guruathwal, 2020-05-13:
> however when using babystep the z-offset is updated to value 0.0 due to the bug reported in #626. There is a missing ":" charater when parsing the echo message. That bug was not yet fixed

@digant73 Have you enabled `DEBUG_LEVELING_FEATURE` in your marlin firmware?

digant73, 2020-05-13:
> > however when using babystep the z-offset is updated to value 0.0 due to the bug reported in #626. There is a missing ":" charater when parsing the echo message. That bug was not yet fixed
> 
> @digant73 Have you enabled `DEBUG_LEVELING_FEATURE` in your marlin firmware?

yes. The problem is that when you execute a babystep command (M290...) you got also an echo response from marlin providing the new z-offset. The parseACK function (in parseACK.c file) does not parse correctly that message due to the bug I reported on #626. The Parser try to convert the value starting with ":" character so the result stored in the z offset variable is always 0.
Below you can see the current part of the code affected by the bug:

else if(ack_seen("Probe Offset"))
      {
        if(ack_seen("Z"))
        {
          setParameter(P_PROBE_OFFSET,Z_STEPPER, ack_value());
        }
      }

While below the version that fixes the bug:

else if(ack_seen("Probe Offset"))
      {
        if(ack_seen("Z:"))
        {
          setParameter(P_PROBE_OFFSET,Z_STEPPER, ack_value());
        }
      }

ack_seen("Z") must be replaced with ack_seen("Z:") (see the presence of ":" character in the fixed version)

Furthermore, in the current bugged version of the TFT FW, the z offset value is restored to the correct value if you for example execute M503. With that command the z offset provided by marlin is correctly parsed and the z offset variable is restored to the correct value. This different behaviour with the M290 is simply due to the different part of code parsing the marlin response (for M290 is bugged for M503 is not bugged)

roomme13, 2020-05-14:
I think Z-Probe offset will not change immediately CMMIW.

guruathwal, 2020-05-14:
@digant73 you need to disable `DEBUG_LEVELING_FEATURE`, then the Z offset will stay as expected. The TFT gets offset settings through **M851** and **M503**

digant73, 2020-05-14:
@guruathwal I understand that disabling DEBUG_LEVELING_FEATURE the problem will not be present but however the parser should be fixed. It's just a missing ":" character. I obviously fixed the bug in my code and tested it. it's working properly

digant73, 2020-08-24:
@SonBroku
The bug was fixed. Can you verify and close the bug?

closed 2020-09-06 by SonBroku