bigtreetech/BIGTREETECH-TouchScreenFirmware #687: [BUG] When printing from USB, printer begins to just randomely air print, skipping all start gcode.

SonBroku, 2020-05-12:
As the title says. If i slice a print in prusaslicer, then put it on a USB stick and attempt to print with it over the TFT, it seems like it just begins air printing and skips a ton of the start gcode. Is this thing not compatible with non-cura slicers?

SonBroku, 2020-05-12:
Yep, it only does this with prints generated with prusaslicer. That's annoying. I don't mind using cura but if I have to use octoprint or the onboard SD to print prints sliced with anything but cura, that's gonna be kind of annoying.

SonBroku, 2020-05-12:
I'm also getting random stops in the print when printing over usb with the TFT....This thing is so buggy holy crap. I'm just gonna use the LCD mode and print with octoprint.

MrRegata, 2020-05-18:
I have the same error with my new tft35 v3.0, and i try to edit the prusaslicer gcode file with notepad++, and cut the first lines where prusaslicer config the accelerations, velocities, etc, and copy after that my start gcode lines. So it´s work for me, when i print from USB, try and comment.

MrRegata, 2020-05-18:
Other way to can print from TFT USB port is change the firmware option to RepRap/Sprinter in config printer/General. This type of firmware is the same as Marlin, but not include the first comands line M20x in gcode file.

This lines produce error when print from TFT USB port.

digant73, 2020-05-22:
please try if with PR #663 the problem is solved