bigtreetech/BIGTREETECH-TouchScreenFirmware #688: [BUG] Prints from onboard SD will not cancel or pause properly from touch menu.

SonBroku, 2020-05-13:
### Description

Whenever I start a print from the onboard SD on my SKR 1.3, and then attempt to cancel said print, I get a black loading screen that stays there forever, while the print continues on. I'm used to having to wait for the buffer to clear when canceling from other print hosts such as octoprint or pronterface, but this seems like it's not working correctly. Pause print also does nothing.
### Steps to reproduce

1. Start a print off "OnBoard SD"
2. Cancel print or pause print from TFT touch menu.

**Expected behavior**

The print finishes the moves in the buffer and then runs the cancel GCODE from the TFT. Or in the case of a pause, the print is paused properly.

**Actual behavior**

When cancelling, A black loading screen appears, the screen becomes non responsive, and the printer has to be power cycled to cancel. If I run the print with octoprint also connected, I can use octoprint to cancel the print, however.

When pausing, the printer simply ignores the command and keeps on.

### Hardware Variant

TFT35 e3 v3.0
### TFT Firmware Version & Main Board Firmware details

Release x.26.1
Marlin 2.0.5.3
### Additional Information

I am using the advanced pause feature with marlin, so I will try turning that off and seeing if the strange pause behavior stops.

Configuration.h:
https://pastebin.com/85LDfJmU

oldman4U, 2020-05-14:
Hi SonBroku:

This should be fixed here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/672

SonBroku, 2020-05-14:
@oldman4U I'm still experiencing this infinite load screen/unresponsive display without prints canceling with these fixes applied to the code and re-compiled and flashed :(. I can cancel TFT prints (when they work and don't hang constantly.) but SD prints don't seem to be getting the M524 at all and the screen just goes into a black loading screen forever. I have to power cycle the printer to stop the print.

oldman4U, 2020-05-14:
OK. Sorry i did not know that you already used the fix.

digant73, 2020-05-14:
> @oldman4U I'm still experiencing this infinite load screen/unresponsive display without prints canceling with these fixes applied to the code and re-compiled and flashed :(. I can cancel TFT prints [...]

As far as I saw from the source code and the tests I made the probems should be related to:
1) start and end gcode are not properly loaded and applied so if you enable them with send_start_gcode and send_end_gcode parameters in the config.ini file the printing menu will not work properly (it doesn't begein the print, no gcode is executed aborting, terminating the print.
2) the cancel gcode is hard coded so the value that you can set on cancel_gcode parameter in the condif.ini file is never applied

Try to disable send_start_gcode and send_end_gcode in the config.ini file (set them to 0 value).
At least you should be able to pause, abort the printing without any black screen, freeze etc...

SonBroku, 2020-05-14:
@digant73 It still does the black screen with no cancel with the start and end gcode disabled in config.ini.

SonBroku, 2020-05-15:
Pausing does seem to work now. It's just canceling that I can't seem to get to work. I updated to the most recent marlin bugfix which seemed to help some behaviour in regards to the master branch firmware with a few fixes applied, but I still can't seem to get an onboard SD print to successfully cancel.

digant73, 2020-05-15:
> @digant73 Pausing does seem to work now. It's just canceling that I can't seem to get to work. I will start a simple calibration cube print from the onboard SD, then immediately cancel it. I'll wait and wait but the screen just stays on the black loading status and the print continues on, forcing a power cycle. If I do this with octoprint also connected, I can send an M524 from octoprint and the print will cancel, so that abord sd print command isn't getting sent, and the menu just hangs on loading..
> 
> I updated to the most recent marlin bugfix which seemed to help some behaviour in regards to the master branch firmware with a few fixes applied, but I still can't seem to get an onboard SD print to successfully cancel.

ok. I don't have onboard SD on my Sidewinder X1 so I cannot test it. I'm using the last Master branch and all the last PR. And I print from SD card or USB stick.

closed 2020-05-18 by SonBroku