bigtreetech/BIGTREETECH-TouchScreenFirmware #689: [BUG] Print from TFT SD without preheat produces instability

cwizard, 2020-05-14:
### Description

If I don't preheat ahead of print job, the gcode of the job is seemingly ignored and the bed and nozzle are not allowed to heat before the job just starts anyways.

### Steps to reproduce

1. Do not preheat
2. Start job from TFT SD

**Expected behavior**

Observe bed heat to 60c and wait
Observe nozzle heat to 200c and wait
Observe print job start

**Actual behavior**
Observe bed start heat to 60
Observe nozzle start heat to 200c
Observe print job start with nozzle at around 100c

### Hardware Variant

TFT35 E3 v3 on master branch - I am recompiling to release branch to see if the behavior exists

master branch & SKR Mini E3 stable branch firmware

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

roomme13, 2020-05-14:
Try this PR #670, it should work

cwizard, 2020-05-15:
Thanks, if I understand what I need to do I cloned a copy of creamywhite's branch.  I will compile it, thanks.

closed 2020-05-24 by cwizard