bigtreetech/BIGTREETECH-TouchScreenFirmware #690: When encoder used in touch mode to adjust values, feed rate [FR] (feature request title)

radek8, 2020-05-14:
**Is your feature request related to a problem? Please describe.**
Hi
I would like to reopen issue https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/680

If it moves with the encoder in touch screen mode, it can change the print speed. Merlin monitors the movement of the encoder in the background and changes the printing speed.

**Describe the solution you'd like**
I have a solution.
Add encoder support to each menu. And add an item to the code:
sendEncoder (2);

For example, add the code to StatusScreen.c

[image]

For example, BabyStep.c

[image]

**This solves the problem when booting to the touch screen.**

But if I switch to the merlin screen and then the touch screen, the problem with the encoder is here again. But why?
I need help with that.

guruathwal, 2020-05-16:
I doubt if this will work properly. what would happen it the knob is rotated to other direction?

radek8, 2020-05-16:
It works.
I've tried it.
You can rotate the encoder in both directions.
You must not press :-) Press will not affect it
You must not switch to merlin mode and back, then the adjustment will stop working.
I do not know why. Probably something will load and remain in memory

closed 2020-05-28 by radek8