bigtreetech/BIGTREETECH-TouchScreenFirmware #692: [BUG] In certain case the file name from the printer's SD card is shown as ???

Tygrys-1, 2020-05-15:
### Description

File named M8x195.gcode is read as ??? on the file listing in touch mode, when the SD card is in the printer slot.

In the Marlin mode the file name is displayed ok.
If the file is placed on the TFT card, the name is displayed ok.
Selecting the "???" file to print starts printing the file, showing the real file name in the upper line of the display.

### Steps to reproduce

1. Name the file on SD card M8x195.gcode
2. Put the SD card in the printer slot
3. Go to touch mode, select print, from printer SD - you should see ??? in the file list

**Expected behavior**

Proper file name should be sisplayed.

**Actual behavior**

??? is displayed

### Hardware Variant

TFT35 E3 v3.0

### TFT Firmware Version & Main Board Firmware details

Marlin 2.0.5.3
TFT 26.1.1, non-modified version

### Additional Information
The file was created in Windows, it has mixed case letters in the name and 6 characters. I suppose that any file name with that properties may have the same problem.

radek8, 2020-05-27:
Build a new firmware. The problem should be fixed.
Do you have long file names allowed in Marlin?
#define LONG_FILENAME_HOST_SUPPORT

Tygrys-1, 2020-05-30:
Sorry to say, it is still the same. Long filenames enabled in Marlin (2.0.5.3) and current (as of 29.5.2020) TFT version installed. 
Still showing ???
Other long and (not so long) names are showing with no problem.

Found another file causing this to happen. It is named A.gcode
What is strange, when I tried to print this file using the TFT mode, on the upper part of the display there is the name A3F7F~1.GCO shown. In Marlin mode it shows and prints ok. File attached.
[A.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4705511/A.zip)
It is actualy not printing anything, so it is safe to try to print.

PR #733 (2020-07-02): Add parsing of fan speed, Improve popup dialog window, Add large fonts, Improve Numpad UI & other bug fixes
- ~Integrate preheat menu with heat menu.~
- Improve Live icons.
- Improve Numpad UI.
- Use Numpad to set temperature instead of increase decrease button. (Click on heat status icons to edit temperatures.)
- Fix 'T0' temperature updating to the current tool/nozzle.
- Fix bugs in baby-stepping and probe offset menu.
- Improve popup dialog window (display window according to message type).
- Fix unnecessary item re-draw in the feature settings menu.
- Add printf/sprintf library from https: [... cut]
closed 2020-07-02 by bigtreetech