bigtreetech/BIGTREETECH-TouchScreenFirmware #693: compilazione config.ini per livellamento 4 angoli in modalità touch

jagger31, 2020-05-15:
salve a tutti ho bisogno di un aiuto nella compilazione del config.ini del firmware di un tft24
il problema è questo:
nella modalità emulatore marlin quando faccio il livellamento dei 4 angoli va tutto bene rispetta tutte le misure di offset del nozzle e distanza del letto dal finecorsa
quando passo in modalità touch e clicco sul livellamento dei 4 angoli del letto mi rispetta le misure le piatto che ho impostato ma non la distanza dal finecorsa all inizio del piatto
ho un letto caldo di 220mm x 220mm che si trova ad X=-7mm e da Y=-13mm dal finecorsa dopo aver fatto homing
dove devo inserire i valori giusti di -7 e -13?
qui?
`level_edge_distance: X20 Y20
level_z_pos:0.2
level_z_raise:10
level_feedrate: X6000 Y6000 Z6000`
qui?
`pause_retract: R15 P16
pause_pos:X10 Y10
pause_z_raise:20
pause_feedrate: X6000 Y6000 Z6000 E600`
qui?
`size_min: X0 Y0 Z0
size_max: X220 Y220 Z200`
o dove?
nel firmware marlin caricato nella scheda bigtrretech skr mini v1.1 è impostato bene ma perche il livellamento del touch non ne tiene conto?
ultima domanda io modifico il config.ini poi lo copio nella scheda sd da 4gb poi la inserisco nell lcd e resetto la stampante ,è giusta come procedura?
oppure nella scheda ogni volta che modifico il file config.ini devo copiare nuovamente anche il file .bin e la cartella dei font?
il config.ini lo modifico o con visual studio e poi clicco su SALVA oppure lo apro con lettore di testo pluma va bene lo stesso?

digant73, 2020-05-15:
Alcune descrizioni sul formato dei parametri che trovi nel file config.ini non sono corrette.
Uno di questi è proprio il parametro level_edge_distance.
Per questo parametro dai il seguente valore:
level_edge_distance:20
dove 20 (millimetri) è la distanza dal bordo del bed sia sull'asse X che Y che verrà usata per i 4 punti.
Ovviamente invece di 20 metti il valore che ti interessa.
Il file config.ini lo metti sulla root SD card e riavvii la stampante per farglielo caricare. Una volta caricato, non ti servirà più usare la SD card con il file nei prossimi riavvi della stampante a meno che non apporti cambi al file e desideri caricare nuovamente le nuove configurazioni nella SPI flash della stampante.
Ad ogni modo, l'introduzione recente della configurazione basata su file è ancora affetta da diversi bug e ha creato svariati  bug che sono in fase di risoluzione nel prossimo imminente rilascio sul master branch.
Il consiglio è di aspettare ed usare la nuova release. Fisserà tutti i bug riscontrati

jagger31, 2020-05-15:
grazie per il chiarimento sul bug ho tirato un sospiro di sollievo credevo di aver fatto casini io.
due soli chiarimenti se posso chiedere 
1 `level_edge_distance:20
dove 20 (millimetri) è la distanza dal bordo del bed sia sull'asse X che Y che verrà usata per i 4 punti.
Ovviamente invece di 20 metti il valore che ti interessa.` 
per quanto riguarda questo bug seguirò il tuo consiglio e aspetto una relase stabilizzata,percio il 20 sta per la distanza del sondaggio all interno del piatto di 20mm percio mi confermi invece che per quanto riguarda la distanza dell inizio del piatto e il finecorsa dell asse x e della asse y rimane impostato quello che ho impostato nel marlin giusto?oppure anche quello va impostato nel firmware del tft24?
2 quando mi dici di mettere il file nell root della sd card mi stai dicendo di copiarla semplicemente come farei con qualsiasi file giusto?(come ho visto negli esempi di carica del file qui su github)o stai parlando di un altro posizionamento che non conosco purtroppo?
ultima cosa posso caricare il file config.ini da solo o per forza insieme al file.bin e alla cartella tft24 delle font come quando faccio il nomrale aggiornamento?perche se carico da solo il file config.ini non vedo nessun contatore di caricamento in uso è corretto che sia cosi giusto?

grazie ancora

digant73, 2020-05-18:
si il level_edge è solo la distanza dei 4 punti durante il manual leveling.
Si il file config.ini lo copi come un normale file
puoi copiare solo il file config.ini senza il .bin e la dir MKS. All'avvio della stampante dovresti vedere un menu di aggiornamento

oldman4U, 2020-06-15:
Hello.

Could you please close this ticket in case you do not need it anymore.

Thank you

oldman4U, 2020-09-05:
@jagger31 

Close this ticket.

oldman4U, 2020-09-11:
@bigtreetech 

Could you please close this ticket.

Thank you

closed 2020-12-15 by stale[bot]