bigtreetech/BIGTREETECH-TouchScreenFirmware #696: [BUG] Knob LED Color

Chris1231, 2020-05-17:
I have updated my TFT35 E3 V3 to the current version (TFT_E3_V3.0.26.1)
but now the colors of the LED Knob are completely gone.
no color is correct.
OFF also no longer works.

sbeck14, 2020-05-20:
Having a similar issue. I copied `BIGTREE_TFT35_V3.0_E3.26.1.bin` and the `TFT35` folder from the Material theme folder to the root of my SD card, and updated the TFT35 E3. When it was finished updating, the LED was stuck at a whitish color. Could not change to any other color or turn it off from the Features menu.

Reverted to BIGTREE_TFT35_V3.0_E3.25.4.bin which still works.

visualcortexlab, 2020-05-20:
same for me.
Even compiling with the latest branch ends up with the same problem. Knob LED always full colour ON.

oldman4U, 2020-05-21:
Hi.

I just updated my TFT35 E3 v3 display to the **master** release, using the config.ini file and the Knob works fine. Still some other minor issues, but it seems that they will be fixed soon.

Tygrys-1, 2020-05-23:
Updated to the master 23.05.2020 and still there is no way to set knob collors. It looks like the problem with the led timing, as the bus speeds were changed in 26.1

oldman4U, 2020-05-24:
I can set the colour of the knob LED using the config.ini file and later using the touchscreen Menu - Settings - Feature. The color is also restored after power off.

TFT35 E3 v3 and SKR E3 DIP.

cwizard, 2020-05-25:
This issue is fixed in the development branch as far as I can tell.  Can we close this issue.

Tygrys-1, 2020-05-25:
> This issue is fixed in the development branch as far as I can tell. Can we close this issue.

I have compiled developement branch (I think called "master") from 25.05.2020 and the problem is still there. Can you point to the working firmware/branch link?

oldman4U, 2020-05-25:
Have you defined the color using config.ini or TFT itself?

Tygrys-1, 2020-05-25:
> 
> Have you defined the color using config.ini or TFT itself? Tygrys-1 <notifications@github.com> 

Tried both. The Knob lights white, but not full bright. It does react on features menu options, nor the config. The config. Was used together with the update, I haven't tried the config by itself.

oldman4U, 2020-05-25:
And you have a TFT35 E3 v3? If so please attach you firmware as well as
your config. ini file here.

Tygrys-1, 2020-05-25:
I have TFT35 E3 v3.0

Firmware is build on today's state of the Master branch, no modifications except setting the board type in platformio.ini
default_envs = BIGTREE_TFT35_E3_V3_0

[BIGTREE_TFT35_V3.0_E3.26.1.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4678620/BIGTREE_TFT35_V3.0_E3.26.1.zip)
[config.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4678622/config.zip)

oldman4U, 2020-05-25:
Hi.

Thank you for the files. I uploaded them to my TFT successfully and the know LED changed from the red I have had before to blue and some other settings also changed and I was able to change the colour of the knob using the menu.

One last question. When you upload a new firmware, do you see the upload process followed by a calibration screen on the TFT and are the two files on the memory card renamed to .CUR?

Tygrys-1, 2020-05-25:
I have this calibration thing after upgrading to 23.05 version, but on the last attempt it was not asking for this. Files are renamed to CUR, when reflashed the fonts, icons and so are showing.
The information screen of the firmware is showing version TFT35 E3 26.1 25.05.2020, so I am sure the firmware is loaded corectly and the right version. I have RCT6 version CPU, no RET6, maybe this is a difference somewhere?

Tygrys-1, 2020-05-25:
Flashed 25.4 and knob LED after the flash had that white color, but then I could use feature menu to change the color.
Flashed 26.1 and stuck on white again.
Tried to change via the config.ini - no change, still white.

I wonder why it work on your's.

oldman4U, 2020-05-26:
Not sure what else could cause this, so not sure what we could test next. I verified your software on my hardware and everything worked as expected. You reported that changing the color works with firmware 25.4 and using this version, you can also change the color of the knob using the TFT. But using the latest version, this all does not work.

I am sorry, but i am out of ideas.;-(

Tygrys-1, 2020-05-26:
> I am sorry, but i am out of ideas.;-(

Thank you for trying and for the time spent on this.

I think of the timing issue. The LED driving code changed from 24.5 to 26.1. Also the bus speeds have been changed and I don't know how this affects timer6 which is driving the LED data line. I was thinking of the error in the new code, but as it is working on your hardware it has to be something more subtle.

The current code seams good, but it might be to fast(rather) or to slow comparing to the older implementation. As I do not know the hardware internals of the board and the F2 CPU, it is hard for me to trace the difference. Maybe there are some different versions of BTT E3 V3, or it is just the tolerance of the WS2812 and some catch up, and some not. Can you suggest changes to the code I can test on my side? Like slowing the LED cycle?

As other people are also affected by this issue it is not connected to my particular board.

drackp2m, 2020-05-26:
Over here another individual with the same problem in the TFT35 E3 V3.0.

It really does show how it makes a tiny attempt to change color, but without a doubt the most annoying thing is not being able to disable it.

I'm going to try going back to the commit of V 35 to see if there's any luck.

A greeting!

Tygrys-1, 2020-05-26:
The current led code implementation looks better than the old one. It just doesen't work well ;)
I think it needs some fine tuning.

cht47, 2020-05-26:
Can confirm that I can't change the color in the menu. I mean something is changing from more white to more blue.. but no yellow etc. its like only white and blue is working partly.. 
Also tried Indigo with config file.. and then switching color in the menu.. blue and white is the only color I get. (or at least it looks like that, my display is in a case, if necessary I can also disassemble it and see which LEDs are not lit)
yesterday I did the pull of the master branch and build it today (TFT35 E3 V3)

oldman4U, 2020-05-26:
OK. I checked my TFT35 E3 v3 again and all colours and OFF work. It is possible to discuss about yellow (a bit green) and orange (not good looking orange) but red, blue, green, indigo and violet
[image]
[image]
[image]
[image]

 are really looking good and they are bright. Checked the firmware and it is 26.1

oldman4U, 2020-05-26:
And this is the CPU on my display.
[image]

drackp2m, 2020-05-27:
> yesterday I did the pull of the master branch and build it today (TFT35 E3 V3)

The latest version of TAG 26 also worked for me (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/Vx.x.26).

In case someone wants to use the latest stable version (regarding LEDs) without compiling anything himself.

> It is possible to discuss about yellow (a bit green) and orange (not good looking orange)

And here it seems to me that they are very right, orange seems yellow and yellow seems green.

Tygrys-1, 2020-05-27:
This version 26.0 (11 Apr) was originally on my TFT and it worked ok, but there the LED code is the same as in 25.4. The change was done in release 26.1 and from that point it stopped working for me. I think it is all about WS2812 tolerance and for some users it fits in, and some are slightly out.

PR #728 (2020-06-01): Restore the ability to change knob colors on TFT35 E3 v3.0, broken in release 26.1
### Description

Changed knob LED timing from 0.4us to 0.35us to correct colors of the knob on the TFT35 E3
According to the WS2812 documentation 0.35us is the right timing to set here.

### Benefits

On some TFT35 E3 boards release 26.1 broke the knob lighting. This is caused by different tollerance of the "0" timing of different WS2812 LEDs. This PR restores the ability to set colors to the knob leds or to switch them off.
 
### Related Issues

Resolves #696
closed 2020-06-01 by bigtreetech