bigtreetech/BIGTREETECH-TouchScreenFirmware #699: 24bit color in simulated mode

mjfox33, 2020-05-20:
It would be nice if the simulated marlin mode used 24bit color instead of the current 16bit. Setting the background and text colors by supplying a RGB hex value in the config.ini file and adding a menu option to set them by RGB hex value when in touch mode would be an added bonus.

Msq001, 2020-06-04:
Hello， #755 has realized yout feature request, you can test it now.

oldman4U, 2020-06-15:
Hello mjfox33.

Have you tried the fix in the meantime?

mjfox33, 2020-06-15:
Thank you for following up! I am not sure how to go about pulling this change in and doing the custom build. Do you have some quick instructions I can follow?

oldman4U, 2020-06-15:
You are very welcome.

Have you been using Visual Studio Code before? And have you ever updated the firmware of your TFT or mainboard?

mjfox33, 2020-06-15:
Yes, I have both the skr miki e3 v1.2 and the tft 35 on my ender 3 pro and I did update them to the latest compiled build when they arrived a few weeks ago. I used the e3 firmware that uses the bltouch for the z stop and the unified menu for the tft35. That part was easy since I just took the binaries already compiled. I did also download visual studio code and did a git clone of the repository and built the master but I think that just did the released build. I'm set up on a MacBook Pro if that helps.

mjfox33, 2020-06-16:
Ok, so I pulled the latest and noticed your changes were already committed to the master branch.  I tried using the following two lines below. It updated successfully but set the foreground and background both to white instead of the colors I had requested. I'm not sure how to help determine where it failed and defaulted to both being white.

# Or set the color(RGB888 format) hex value directly(start with “0x”).
# Such as: 0xFF0000 : Red, 0x00FF00 : Green, 0x0000FF : Blue
marlin_bg_color:0x002b36
marlin_fn_color:0x839496

oldman4U, 2020-06-17:
I can see you already posted this problem in #755 . Lets see what the answer is.

oldman4U, 2020-09-05:
Could you please test the latest build and check if it works.

Thank you

oldman4U, 2020-09-11:
Any update on this?

mjfox33, 2020-09-11:
This can be marked as completed.

oldman4U, 2020-09-12:
Then please close the ticket. You are the only one who can do this.

closed 2020-09-14 by mjfox33

oldman4U, 2020-09-14:
👍🏻