bigtreetech/BIGTREETECH-TouchScreenFirmware #700: Bigtreetech SKR E3-DIP v1.1 and Bigtreetech TFT24-v.1.1

YapCK45, 2020-05-21:
I bought a Bigtreetech SKR E3-DIP v1.1 and Bigtreetech TFT24-v.1.1. According to the documentation it should be fairly straight forward to get them to work. I connected them as in the photo below:
[image]
 
After powering up it shows the both marlin and touch mode:
[image]

But it does not respond to the rotary switch. I am new in this and would appreciate any advice and troubleshooting tips.

YapCK45, 2020-05-21:
Checked with Creality 3D® 12864 LCD Display Screen and it works:
[image]

radek8, 2020-05-21:
I assume you have a connector configured as EXP3 on the SKR-E3 board, but you are connected to the EXP1 connector on the TFT.
The connectors do not have the same pin layout. The encoder pins are in the EXP2 connector

radek8, 2020-05-21:
Wrong display selection.
The TFT35-V3.0-E is a direct replacement for the Ender3 display

YapCK45, 2020-05-21:
Would you be able to provide the pin connection for EXP1 connector on the TFT to the SKR-E3 board. I think that's what I need.

closed 2020-05-21 by YapCK45