bigtreetech/BIGTREETECH-TouchScreenFirmware #702: Button locations are incorrect after flashing

xxKeoxx, 2020-05-21:
I was running my Bigtree Tech TFT35 V3 with the screen flipped and flashed the newest firmware.  After flashing the newest firmware all of the icons appear to be in the correct place but the result of the icon press seems to be reversed.  For instance, the back function responds when I press the upper left icon when the actual back icon is located in the lower right.  I've tried to re-flash the previous version and it still has the same problem now.  I've tried to use the reset.txt file in the SD card root, but when I attempt to calibrate the touch screen it keeps saying failed please try again.

I am out of ideas on how to fix this.  Is there anyway to just reset everything on the screen so no firmware exists?

Or better yet, does anyone know how to fix this issue?

oldman4U, 2020-05-22:
Hi.

I am running a TFT28 with flipped screen and it works, even the screen is flipped while upgrading the FW. Have you seen the flip option in the config.ini which has to be turned on (1).

Hope this helps

oldman4U, 2020-06-08:
Hi @xxKeoxx.

What is the current status of the reported issue?

Please let us know

xxKeoxx, 2020-06-08:
I am so sorry.  I thought I responded to this.  I am still having the issue but I was able to work around it.  So after I flashed the firmware all of the icons were correct but the actual button placement was upside down.  I had to try to navigate the menus upside down from memory.  Once I got to the menu I rotated the screen back to the original origin and then switched it back to the flipped origin.  After that it functioned as expected.

However, when I flashed the firmware again I still had the same problem that I initially reported.  If I flash the firmware with the screen in the upright orientation, everything works as expected.

oldman4U, 2020-06-09:
I use a TFT28 flipped, so i can only speak for this kind of display. Updating the firmware, the screen is flipped during the update, but is correct after that. Have you tried to use Master Branch instead of v25.x and have you tried to edit the config.ini file, where you can activate the upside-down screen:

#### Rotate UI 180 degrees
# Options: [enable: 1, disable: 0]
rotate_ui:1

oldman4U, 2020-06-15:
Hi.

Could you please close this ticket in case you do not need it anymore.

Thank you

closed 2020-07-04 by bigtreetech