bigtreetech/BIGTREETECH-TouchScreenFirmware #704: [BUG] Cannot exit print menu after finish Onboard SD print, ABL menus not shown after reboot, Etc..

roomme13, 2020-05-22:
### Description
I encounter some issue here.

-After finish print from Onboard SD, i cannot go back to main menu, if i press back button. it show loading, but never reach to main menu until i reset the LCD.

-Always stuck in loading screen when you open Onboard SD menu if SDCard not available at that time rather than shown "No SDCARD" error messages.

-ABL menu will not shown after reboot /reset LCD, must re enter M115 again, but work normally when i left SDCARD Onboard in place.

-EEPROM Restore (M501) in custom GCODE not show any messages when pressed, so i dont know it really restored or not

### Hardware Variant
TFT35 V3.0
SKR Mini E3 V1.2

### TFT Firmware Version & Main Board Firmware details
Latest build from master with #663 
https://github.com/roomme13/BIGTREETECH-TouchScreenFirmware/tree/build-xx.26.1

### Additional Information
I use latest Marlin Bugfix2.0.x 
https://github.com/roomme13/Marlin/tree/build-bugfix-2.0.x
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

roomme13, 2020-05-23:
Update with latest commit in master, still have same issue.
i already reflash config.ini and reset EEPROM with reset.txt several times.

by the way out of topic, how can i get serial output from LCD to PC /Mac?

radek8, 2020-05-25:
> 
> by the way out of topic, how can i get serial output from LCD to PC /Mac?

I solved it via the ESP-01S Wifi card. Connected to TFT display.
I have a "HW Virtual Serial Port" program running on my PC, which maps the IP address of the Wifi card

roomme13, 2020-05-31:
@radek8 I have that ESP connected to, currently have ESP3D installed, should i install different firmware to use "HW Virtual Serial Port" in that ESP?

BTW i i see #725, did you face any other issue like a state above?

radek8, 2020-05-31:
> BTW i i see #725, did you face any other issue like a state above?

Sorry, same problem.

> @radek8 I have that ESP connected to, currently have ESP3D installed, should i install different firmware to use "HW Virtual Serial Port" in that ESP?
> 
No other firmware is required.

[image]

[image]

[image]

roomme13, 2020-06-08:
@radek8 Thank you Dude! really appreciated it

Bit update for this Issue, Still same issue from latest commit. 
-Stuck screen after print from OnboardSD
-Stuck screen when open file from OnboardSD when the sdcard no present. (havent try this #753)
-ABL menu will not shown until manually M115
-Most of pop notification missing after lastest commmit. (ABL, Zoffset, etc..)

and i experience some freeze randomly, when navigating thru menu.

oldman4U, 2020-09-11:
Hello @roomme13.

There are several fixes available for the issues you reported. Have you tried the last firmware to see how it works?

roomme13, 2020-09-29:
Hi @oldman4U 
I left for a while and kind of lost with latest update. (even marlin improve so fast)
but i manage to test and seems everything fine for now. 

i'll close this issue

closed 2020-09-29 by roomme13