bigtreetech/BIGTREETECH-TouchScreenFirmware #706: SKR Mini E3 v1.2 with TFT35 E3 v3 wont print from usb stick

lumpy1965, 2020-05-23:
just installed a SKR Mini E3 v1.2 & TFT35 E3 v3, both updated with latest firmware from github.

no Bltouch installed.

the gcode prints fine from the sd card slot, but same file wont print from the usb stick hanging off the side of the screen, as soon as i try the screen flashes over n over with "cold extrusion prevention" error, and then the "X - Y" axis does a jig in middle of the bed then X goes to home.

thoughts?

cwizard, 2020-05-25:
Can confirm this happened to a user I was helping, but cannot explain why.  We used identical firmware and I was not able to reproduce this behavior from my TFT E3v3 when printing from the USB slot.  Printing from the full size SD slot on the TFT works as expected.

Mowhok, 2020-05-27:
how to fix your problem: go into you touch mode. into settings and enable start print and end print gcode. then you can print

cwizard, 2020-05-27:
> how to fix your problem: go into you touch mode. into settings and enable start print and end print gcode. then you can print

This makes little sense, because the same file printed from the SD slot on the TFT prints correctly.  Expected behavior from start/end GCODE settings are that they are additional.  Several users have reported this behavior in the BIGTREE user group on Facebook.

oldman4U, 2020-10-06:
What is the status of the reported issue. Please let us know

closed 2020-12-15 by stale[bot]