bigtreetech/BIGTREETECH-TouchScreenFirmware #708: [BUG] Cannot select the interface version (Marlin/TFT) with the knob.

Tygrys-1, 2020-05-23:
### Description

Cannot select the interface version (Marlin/TFT) with the knob. By pressing the knob I can enter the interface selection screen, but here only touch selection is possible. Previous version have the ability to also use knob for this.

### Steps to reproduce

1. Long press the knob.
2. Rotating the knob does nothing on the selection screen.

**Expected behavior**

Rotating the knob should hilight interface icon.

**Actual behavior**

Rotating the knob does nothing on the selection screen.

### Hardware Variant

BTT 35 E3 v3.0

### TFT Firmware Version & Main Board Firmware details

Marlin 2.0.5.3, BTT 26.1 and master from 23.05.2020

### Additional Information
Stock configuration. I saw this working on YT on earlier versions of the BTT software.

Hans007a, 2020-05-24:
I have this issue too.

Hardware Variant
BTT 35 v3.0
TFT Firmware Version & Main Board Firmware details
Marlin Bugfix-2.0.x, BTT master from 23.05.2020

DrDrache, 2020-05-24:
I also have this problem, my screen is stuck on the Interface Selection screen.
BTT TFT35-V3.0 - Firmware 3.0.26.1
SKR 1.4TURBO Marlin 2.0.5.2

cwizard, 2020-05-25:
Can confirm that using the push knob to get to the UI selection works, but the knob rotation to select Marlin vs BTT Touch interface does not work.

Workaround:  Touch the screen

DrDrache, 2020-05-25:
> Can confirm that using the push knob to get to the UI selection works, but the knob rotation to select Marlin vs BTT Touch interface does not work.
> 
> Workaround: Touch the screen

The workaround does not work in my case - screen is stuck on UI selection even after moving to a new board.

radek8, 2020-05-25:
> > Can confirm that using the push knob to get to the UI selection works, but the knob rotation to select Marlin vs BTT Touch interface does not work.
> > Workaround: Touch the screen
> 
> The workaround does not work in my case - screen is stuck on UI selection even after moving to a new board.

Do you have a TFT display connected via the EXT3 connector?

DrDrache, 2020-05-25:
> Do you have a TFT display connected via the EXT3 connector?

That is a very good question. But no, EXT3 is not a port used in either of my configurations (SKR1.4 and SKR1.4T)
To further clarify - it is connected via :  
EXT1
EXT2
TFT rs232 port.

radek8, 2020-05-25:
Can't you have the same problem with the cover as here?
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/558

cwizard, 2020-05-26:
Yeah I would like to know if the behavior persists when removing the TFT from the chasis

Tygrys-1, 2020-05-29:
> 
> Yeah I would like to know if the behavior persists when removing the TFT from the chasis

AFAIK mine issue is not connected with the touch screen, but with the lack of ability to use knob on this selection screen. The touch "workaround" works good for me.

oldman4U, 2020-07-07:
Hello Kitty. Its me. Oldman.

Could you please close your tickets which you do not need anymore.

Good cat;-)

Tygrys-1, 2020-07-07:
I do not think this problem has been solved. The knob is not moving the choice.

oldman4U, 2020-07-07:
I believe this will not change any more.

As far as I know, the ability to change to mode by rotating the encoder has been removed, because rotating the encoder also changed the FR in Marlin automatically. So to prevent the speed to change by rotating the encoder, the ability to select the mode with the encoder has been removed. Which is "a problem" for those who know that this has been possible with an earlier version, but is no problem for new users.

But it your ticket and of course you can keep it opened if you think it makes sense.

I believe it would be good to have only tickets opened which are important, so there is one or two pages of tickets, but now more

Thank you

traffic-light, 2020-09-12:
@bigtreetech this one can be closed

oldman4U, 2020-10-06:
Pussycat. I hope you are ok. Haven't heard from you since a while. Could you please close this ticket as the reported issue is solved.

Thank you

closed 2020-10-07 by Tygrys-1

oldman4U, 2020-10-07:
Everything ok Tiger!!??

oldman4U, 2020-10-07:
We miss you!!!🥺

Tygrys-1, 2020-10-07:
Thanks for remembering me ;-) Just taking other activities at this time.

oldman4U, 2020-10-07:
Of course, we miss you here. But good to hear you are ok. You never know these days...

👍🏻