bigtreetech/BIGTREETECH-TouchScreenFirmware #709: [BUG] Emulation mode no longer works with the latest update

GenaUser, 2020-05-24:
### Description
Emulation mode no longer works with Repetier firmware.
@bigtreetech, A few weeks ago, you have restored this compatibility for the touch mode. It is lost again, but now in emulation mode.
The new detection of Mainboard firmware type is probably the cause. Marlin, Smoothieware, but not Repetier.
Please, can you correct this problem (and the connection bug #552 ).
If you no longer want your screens to be usable by Repetier users, can you confirm this?
We need clear information to be able to continue using your products.
Thanks in advance

### Steps to reproduce
See the video:
https://youtu.be/zfRBMia7DzE

### Hardware Variant
TFT35 V3

### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT35_V3.0.26.1 / RUMBA+ and RAMPS / Repetier 1.0.3

### Additional Information

bigtreetech, 2020-05-25:
Hello , the issue of Emulation mode has be fixed in #714 ,  and #552 seems not a bug of TFT, I tested it on the BTT Base V1.4 board and flash Repetier-Firmware v1.03. Everything is normal !

Please connect your motherboard to the computer using USB, and check whether the motherboard feedback info is normal when reset by printrun(or some other monitoring software). And attach the log in here

GenaUser, 2020-05-25:
Hello @bigtreetech, the emulation mode works now. Thanks.
On the other hand, the #552 is not completely solved. After update, everything is normal... until reboot...
I made a small video for you. Please take a minute to watch the problem: https://youtu.be/zfRBMia7DzE
I can't wait to reuse your screens (I have a big family of ducks...)

GenaUser, 2020-05-25:
Here is the log file:
  12:47:56: Attempting to connect to printer
  12:47:56: Connection opened
< 12:47:57: start
< 12:47:57: Detected EEPROM version:19
< 12:48:02: Free RAM:3419
< 12:48:02: SelectExtruder:0
< 12:48:02: FIRMWARE_NAME:Repetier_1.0.3 COMPILED:Apr  9 2020 FIRMWARE_URL:https://github.com/repetier/Repetier-Firmware/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:Mendel EXTRUDER_COUNT:2 REPETIER_PROTOCOL:3
< 12:48:02: Cap:PROGRESS:1
< 12:48:02: Cap:AUTOREPORT_TEMP:1
< 12:48:02: Cap:EEPROM:1
< 12:48:02: Cap:AUTOLEVEL:0
< 12:48:02: Cap:Z_PROBE:0
< 12:48:03: Cap:SOFTWARE_POWER:1
< 12:48:03: Cap:TOGGLE_LIGHTS:1
< 12:48:03: Cap:PAUSESTOP:1
< 12:48:03: Cap:PREHEAT:1
< 12:48:03: Printed filament:756.49m Printing time:7 days 5 hours 15 min
< 12:48:03: PrinterMode:FFF
< 12:48:03: Unknown command:M503  S0
< 12:48:04: X:0.00 Y:0.00 Z:0.000 E:0.0000

Msq001, 2020-05-25:
Hello, Can you provide the ".hex" file of firmware or source code you use so that I can use the same environment for testing?

bigtreetech, 2020-05-25:
Hello, Can you provide the ".hex" file of firmware or source code you use so that I can use the same environment for testing?

GenaUser, 2020-05-25:
Can you download this firmware from OneDrive?
https://1drv.ms/u/s!Ak8QYQq1MpUzmGeW-NkxIpEXMFnX?e=uRcJGX
Or do you prefer another method?

bigtreetech, 2020-05-27:
Hello, #552 should be fixed in #719.
It‘s because of mega2560 will be hanged when received serial data at startup.
now the TFT will send query temperature gcode 1 second after startup not in startup, so it should work fine now.

GenaUser, 2020-05-27:
Hello, the problem seems solved.
Thanks a lot for your job.

closed 2020-05-27 by GenaUser