bigtreetech/BIGTREETECH-TouchScreenFirmware #711: [BUG] (TFT35 status bmp names incorrect for 26.1 on TFT35_V3.0_E3.26.1)

raf802, 2020-05-24:
### Description
I noticed that flashing the new firmware with unified menu theme that a few icons were not found while flashing. e.g. Nozzle Status
Issue persists with both firmware versions on github. 

The issue does **not** exist in the x.x.26.1 branch.

The issue is due to the names for the nozzle, bed & fan status are incorrect. The bmps are named "Status_Nozzle", etc in the master branch for TFT35 however the name which the firmware is looking for is "StatusNozzle" which they are called in the other branch. 

I have not checked the other TFT versions.

### Steps to reproduce

1. Drag-drop TFT35 folder from master branch to sd card root folder
2. Follow flashing procedure as normal

**Expected behavior**
All bmps are found and flashed correctly

**Actual behavior**
Status icons for the nozzle, bed and fan are missing and show a different icon.

### Hardware Variant
BTT TFT35 E3 V3

### Additional Information

cwizard, 2020-05-25:
This isn't a bug.  There are differences and new icons in the development branch (master) and you will need to take note of those, or use the icons provided in the development banch.

cskozlowski, 2020-05-29:
There are 3 bitmap (bmp) files that are incorrectly named:  Status_Fan.bmp, Status_Nozzle.bmp and Status_bed.bmp.  I copied these three files and gave them the following names:  StatusFan.bmp,StatusNozzle.bmp and StatusBed.bmp.  I then copied the appropriate *.bin file and bitmap/font file (in my case in was TFT35.  This solved the problem of not finding the 3 bitmap files for me.  Hope this helps.  I also see lots of complaints of 'bricking' this display during a firmware load,  erase your SD disk and reformat it - then load the three elements (BIGTREE_TFT35_V3.0_E3.26.1bin, TFT35 directory, and config.ini) on the fresh SD card - that has solved many issues for me.

oldman4U, 2020-06-01:
Hello raf802.

Could you please test the above and CLOSE this ticket, Thank you.

Maybe you have not seen this beautiful theme made by @Acenotass: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/681

This also works with the latest Master branch.

closed 2020-06-01 by bigtreetech