bigtreetech/BIGTREETECH-TouchScreenFirmware #713: [Q] TFT35 V3.0 SW26.1 > Communikation on RS232 change to which port after received first answer "T:50"?

Wesie, 2020-05-25:
I'm a new owner of the TFT35 V3 and I have flashed the latest SW 26.1. 
The Display is in this moment connected via an RS232>USB adapter at the PC with a running terminal console. (GND, 5V, RX, TX) with the black 5 wire cable (incl RST)

I've recognised, that the "M105" message polls and the TFT is waiting for a response => OK
The information message that the Printer is not connected is visible => OK
If I send an answer like "ok T:50" to the Display, the Display accept the answer. The missing printer informations is cleared. => OK

But up from this moment there is no communikation anymore on the serial RS232 Port possible.
Is this behaviour correct? 
Which serial port is now active?
I would expected, that the RS232 port will be still acttive, if the TFT received the message on this port. Isn't it?

bigtreetech, 2020-05-25:
When connected to the printer, the TFT will not send M105 out continuously. The TFT will send the next Gcode only after the previous Gcode receives the correct response ("ok \ n")

Wesie, 2020-05-25:
I've found the reason/solution:
In the Configuration.h there is an option
/** * Keep Serial always On (ONLY SUPPORTED ON TFT24 V1.1, TFT35 V3.0, AND
TFT28 V3.0)
* Keep UART (Serial communication) alive in Marlin Mode *
* Allow seamless OctoPrint UART connection to the TFT's UART/serial
expansion port no matter which mode the TFT is in. *
* Options: 0: Disabled 1: Enabled */
#define SERIAL_ALWAYS_ON 0 // Default: 0 (Disabled)

It should be set to 1

closed 2020-05-28 by bigtreetech