bigtreetech/BIGTREETECH-TouchScreenFirmware #715: [BUG] TFT35-V3 Touchmode to SKR1.4Turbo on Marlin 2.0.5.3 NO PRINTER ATTACHED

Red-Devil74, 2020-05-25:
### Description

My tft35 v3 screen in touchmode keeps stating "Printer not attached". 
I have configured marlin with SERIAL_PORT 0 an SERIAL_PORT_2 on -1, baudrate 115200. 
I have connected the screen with EXP1 and EXP2 and the TFT cable connecting the 5v pins tx to Rx and Rx to tx, grnd to grnd and rst to rst. I also have toggled the baudrate on the TFT screen to 250000 and 115200.

I have the usb cable of the skr1.4 turbo connected to a RaspberryPi 4 which runs Octoprint. 
If I tell Octoprint to connect te the printer, (baudrate auto discovered 115200) the message "No printer Attached!" disappears and temp info of the printer can be seen on the tft. 
Sending data from the screen (Movements or temp or printing) is not received by the board and printer is not responding

I have updated the firmware of the TFT and run marlin 2.0.5.3. I also recompiled the firmware with various setting with no avail...

### Steps to reproduce
1. Upon boot/reset, Marlin mode works as expected. Touchscreenmode states:"No printer Attached!"
2. Tel Octoprint (via webinterface) to connect to printer
3. "No printer Attached!" message disapears but touchscreen commands are not getting through to the printer

**Expected behavior**
Upon "Home", Printer should do a G28 (as an example)

**Actual behavior**
Nothing happens...

### Hardware Variant
BTT TFT3.5 V3
SKR1.4 turbo
### TFT Firmware Version & Main Board Firmware details
TFT: 3.0.26 (both unified, classic and own compiled version tried)
Marlin 2.0.5.3 (Also tried 2.0.3)

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.
[Marlin and TFT conf files.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4676778/Marlin.and.TFT.conf.files.zip)

radek8, 2020-05-25:
I would try to connect the RX TX cables in reverse

Red-Devil74, 2020-05-25:
@radek8 Thank you for your reply. Regretfully I tried all that (and now agian to be sure). I have even crimped on new connectors to the cable measured the connectors on continuity etc. If I revers RX and TX on one connector (connecting TFT RX to SKR14 RX etc) my octopy can't connect anymore, stating (from serial log):
`2020-05-25 15:20:21,112 - Changing monitoring state from "Operational" to "Offline (Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2823)"
2020-05-25 15:20:21,120 - Connection closed, closing down monitor
2020-05-25 17:41:56,730 - Changing monitoring state from "Offline" to "Detecting serial port"
2020-05-25 17:41:56,757 - Serial port list: [u'/dev/ttyACM0']
2020-05-25 17:41:56,757 - Connecting to: /dev/ttyACM0
2020-05-25 17:42:06,836 - Changing monitoring state from "Detecting serial port" to "Error: Connection error, see Terminal tab"
2020-05-25 17:42:06,880 - Unexpected error while connecting to serial port: AUTO IOError: '[Errno 5] Input/output error' @ comm.py:_openSerial:2691 (hook default)`

cwizard, 2020-05-25:
The RS232 cable needs to be plugged into "TFT" on the SKR motherboard and "RS232" on the TFT otherwise it will not work properly.  Can you please provide photographs of how you have the RS232 cable connected between the SKR and the TFT, this will help others troubleshoot.

Red-Devil74, 2020-05-25:
Sure! I highly appreciate the assistance. See pics attached
[image]
[image]

radek8, 2020-05-25:
Only connect RS232. Disconnect the EXT1 and EXT2 cables.
It will work?

Red-Devil74, 2020-05-25:
> Only connect RS232. Disconnect the EXT1 and EXT2 cables.
> It will work?

I Tried that but doesn't change anything.
If OctoPi is connected and the "No printer attached" message disapears, I can see the temperature of the print-bed and nozzle are shown which led me to think that it receives data. I can not read the "Eeprom" info and I can not Send data,...

giorgakos, 2020-05-25:
No printer attached, in my case also with same configuration 
SKR1.4Turbo on 2.0.5.3 with TFT35_E3_V3.0.26 firmware

radek8, 2020-05-25:
I don't know if it will help, but you have an old version of Marlin.
Install Ma Marlin bugfix version 2.0.x

Red-Devil74, 2020-05-25:
Thank you Radek,
I have tried with Marlin 2.0.3, the current release 2.0.5.3 and I have tried the latest (last weekend) bugfix nightly. I could not see any different behviour, regrettably. 
Could it be a hardware failure?

radek8, 2020-05-25:
This may be a defect in the HW.
I have a SKR1.4 board and a TFT35_E3_V3.0 display
And it works on the first connection

kkosa1, 2020-05-25:
i am curious what setting you are using for   #define SERIAL_PORT_2

Red-Devil74, 2020-05-25:
As stated in my intro, I have:
`SERIAL_PORT 0
SERIAL_PORT_2 -1
baudrate 115200.`

I also have experimented with a lot of combo's like
`SERIAL_PORT -1
SERIAL_PORT_2 0`

`SERIAL_PORT 0
//SERIAL_PORT_2 ` (so no serial port 2)

Just SERIAL_PORT -1 etc,...
Frankly, I am at a loss,.. 
If it is a hardware failure, i still do not know whether it is the TFT or the SKR...

radek8, 2020-05-25:
-1 is for USB
0 is for TFT Connector
3 is for Wifi connector, but you need to adjust the Visual Studio configuration

kkosa1, 2020-05-25:
When I set  #define SERIAL_PORT_2   to 1 in Marlin my "NO PRINTER ATTACHED" problem went away.  i have my ESP 01S WiFi adapter installed on the TFT 35 V3 with no additional WiFi configs in Marlin.  Has been working fine with various Bugfix versions from mid March 2020 to todays version.  Not sure if this could help.

cwizard, 2020-05-25:
I use
#define SERIAL_PORT 2

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 -1

Ender 3 SKR Mini E3 V1.2 & TFT E3 V3

From what I can see, physical connection is proper.  Either firmware misconfig or hardware failed :(

Red-Devil74, 2020-05-26:
@radek8 Thanks for the info
@kkosa1 Do you have SERIAL_PORT defined? and if so at what port?
I tried setting it to one (if I recall correctly) and it gave me an pin error on the LCD screen..
@cwizard So if I understand correctly, you have both ports defined and SERIAL_PORT = 2 and SERIAL_PORT_2=-1. So you have the second port for your USB connection. and your first one for serial2? If I dive deeper in the code (and by no means am  I a programmer,... I just read stuff) I see that serial0 is connecting pins 0.2 and 0.3 (RX and TX, i.e. the rs232 connector on the skr14),... so if I use serial 2, which pins should I then connect?

Thanks all for your help!

regards,
Arthur

radek8, 2020-05-26:
Attention!
The port number depends on the Marlin board used.
Kkosa1 and Cwizard have a different board with Marlin than you!

Red-Devil74, 2020-05-26:
Ah, @radek8 that is quite some usefull info ;-) 
!

radek8, 2020-05-26:
use
SERIAL_PORT -1 
SERIAL_PORT_2 0
or
SERIAL_PORT 0 
SERIAL_PORT_2 -1

radek8, 2020-05-26:
I am based on the SKR1.4 configuration. - 1768 processor
You have SKR1.4Turbo. - processor 1769
In the evening, I'll see if the Turbo board doesn't use other ports, for example

Red-Devil74, 2020-05-26:
Thanks @radek8 that would be nice. As I am not a programmer, I find it hard to change the right files. I was thinking of changing hardwareserial.h, adapt serial0 to refere to the pins of thewifi connector and that attache the RX and TX pins to the other connector but I do not know if it is that simple (currently I am also at Work but find myself easilly distracted... ;-)

radek8, 2020-05-26:
SKR1.4 has the same pin layout as SKR1.4Turbo.
I don't see why it doesn't communicate with the display

Red-Devil74, 2020-05-26:
So I suspect a hardware error... I am checking if I can use the rx and tx pins of the WIFI board (I think USART5?) and remap those to rx and tx of RS232 (ie replace the PA3 and PA4 pin with PC12 and PD2 in UART.c file,... ) A lot of problems,... I would have liked a product that worked out of the box... (and still, the problem could be in the SKR board and then I doné know anymore,...)

radek8, 2020-05-26:
284/5000
If you think there may be faulty ports on the TFT connector, you can use RX and TX from the WiFi connector.
You set up Marlin
SERIAL_PORT -1
SERIAL_PORT_2 3
Before compiling Marlin, you need to edit the HardwareSerial.h file as shown
Attention The WiFi connector has a 3.3V power supply and must not be connected to 5V

[image]

Red-Devil74, 2020-05-26:
@radek I have compiled marlin with serial_port_2 3 an reflashed the skr14, then I changed HardwareSerial as above. Do I now need to reasign the ports in uart.c inside the touchscreen firmware (src/User/Hal/stm3210x)? before I recompile the screen firmware?
thanks in advance!

Red-Devil74, 2020-05-26:
Oh,.. I think you mean that I use the WIFI connector on BOTH SKR as well as TFT...?  I was thinking of leaving Marlin and the skr14 alone and only change tf35 pin and firmware so to check if it is the tft or the skr that is faulty...

Red-Devil74, 2020-05-26:
So I did 2 tests. Here is what I did for test 1:
1. read your post above again,...
2. changed marlin and hardwareserial accordingly
3. compiled marlin and flashed skr14
4. took out 2 connectors from tft cable (RX and TX) and inserted them in 4.29 and 4.28
5. Boot 
5. screen lights up with "No Printer Attached" message
6. Connected the raspberry pi to skr14 via webinterface
7. message disapears (confirming that the pin swap works
8. regretfully no change in reaction from the printer (i.e. problem not resolved)

radek8, 2020-05-26:
What I sent above is a change to the Wifi connector only for SKR1.4

radek8, 2020-05-26:
Are you dealing with the connection of a TFT display or a raspberry pi?

Red-Devil74, 2020-05-26:
test 2:
1. undo change in hardwareserial and configuration.h (ie go beck to SERIAL_PORT_2 0)
2. Compiled and flashed SKR14
3. Set pins of skr14 connection back to original pins
4. extracted pins RX and TX on TFT side and connected these to the WIFI port on tft board
5. edited UART.c on tft firmware so the ports of USART1 and USART5 are swapped (both in TFT\src\User\Hal\stm32f2xx as in TFT\src\User\Hal\stm32f10x because I have no clue what I am doing...
6. flashed the TFT and booted
7. no connection with tft to skr14
8. swapped RX and TX pins (just to be sure)
9. no reaction (and the USB connection of the RPI to SKR14 went nuts....

So the steps here from step 5 onwards are a wild guess..

Red-Devil74, 2020-05-26:
@radek8 I am dealing with the connection of the TFT35 to an SKR14. 
BUT
I also have an RaspberriPi connected to the USB of the SKR14.
Upon normal SKR14 boot, I see a message on the TFT (connected to the SKR) "No Printer Attached" which disappears when I make a serial connection on Octopi on the Raspberrypi with the SKR14... I do not know why this happens,....

radek8, 2020-05-26:
yes, the raspberry pi is connected via USB.

Red-Devil74, 2020-05-26:
So my test one seems to suggest the the error is most likely on the TFT, right? I would have hoped that test 2 could confirm that but I do not know how to do that test propperly (do I need to change hardwareserial.h for that?

radek8, 2020-05-26:
check the display to see if you are using the same port speed as you set in Marlin. 115200?

radek8, 2020-05-26:
You can leave Hardwareserial.h changed.
When you want to use WIFI

Red-Devil74, 2020-05-26:
speed is chacked and at 115200
any ideas on how to swap the ports properly in the display side? I assume wifi is on usart 5?

radek8, 2020-05-26:
I've never redirected the ports on the display, so I don't know
I have a TFT35-E3V3.0
There is
UART1 - WIFI
UART2 - RS232 connector

radek8, 2020-05-26:
It will be the same on the TFT35-V3.0

radek8, 2020-05-26:
if you want to redirect the port to TFT, you should do it in the file - pin_TFT35_V3_0.h

radek8, 2020-05-26:
[image]

Red-Devil74, 2020-05-26:
Many Thanks @radek8 . I will check tomorow and provide feedback!

Red-Devil74, 2020-05-26:
@radek8  Dear Sir, You are a Genius! With the mod on the TFT35 side and recompiled TFT firmware All works as should be making use of the RX an TX pins of the Wifi connection! I now ill ask BTT for a replacement board! Thank you for your assistance!. I do not know how to change this bugreport to a question i.s.o. a bug but I will close this shortly!

radek8, 2020-05-26:
Congratulations.
If they do not accept the complaint, redirect RS232 to UART3.
The WiFi connector will remain free for the WiFi card.

Red-Devil74, 2020-05-26:
Very nice suggestion (use TX3 and RX3). Eventually if they do not react I might do that!. I do think that they shoul at least try to sent a new board or compensate me... Let's see how they respond!

radek8, 2020-05-26:
OK
Good evening.

closed 2020-05-27 by Red-Devil74