bigtreetech/BIGTREETECH-TouchScreenFirmware #717: [BUG] (short description)

rudihorn, 2020-05-25:
### Description

TFT firmware is unable to send emergency parser codes.

### Steps to reproduce

1. Compile marlin with emergency parser support https://github.com/MarlinFirmware/Marlin/pull/18095
2. Issue `M109 S180` from an external program. Note that you cannot send it from the TFT, as the TFT changes `M109` into `M104`.
3. Wait for the TFT to send an `M105` to update temperature.
4. Either press emergency stop or send `M108 g code`.

**Expected behavior**
If you skip step 3. by quickly sending the gcode then it sends the emergency code and it works. 

**Actual behavior**
The TFT waits for an `ok` response.

### Hardware Variant
TFT 35 V2.0

### TFT Firmware Version & Main Board Firmware details
SKR Pro V1.1 with https://github.com/MarlinFirmware/Marlin/pull/18095
TFT Firmware version V2.0.26.1

### Additional Information
This is related to #518. Using `M105` instead of `M155` is just bad, especially with the synchronous command queue nature that the TFT uses. Marlin describes this approach as `less than optimal` and recommends using `M155` which is possible.

I'm also not a fan of the silent changing of `M109` to `M104`.

guruathwal, 2020-05-26:
There were some changes to EMERGENCY_PARSER settings because of issue #695 raised by @thisiskeithb.

thisiskeithb, 2020-05-26:
> There were some changes to EMERGENCY_PARSER settings because of issue #695 raised by @thisiskeithb.

`EMERGENCY_PARSER` support for `STM32` was just merged in Marlin (https://github.com/MarlinFirmware/Marlin/pull/18095), but I'll need to some more testing.

rudihorn, 2020-05-27:
@thisiskeithb the following boards use `STM32` and not `STM32F4` HAL by default though:
* SKR Pro 1.1
* GTR 1.0
* BTT002 1.0

It is only the mini boards that still use `STM32F1`.

rightawy, 2020-06-01:
Earlier a message was displayed on screen that emergency gcodes will not work if marlin did not have emergency_parser enabled but now BTT has removed it. 
earlier this lets us know in advance whether the stop button will work or not on TFT in case of actual emergency. 
Now you may have to learn this hard way that the emergency gcodes will not work, like my cooling fan melted while I was pushing stop button on TFT repeatedly on one of my printer.
@bigtreetech should put that message back, to let know that the emergency gcodes will not work.

closed 2020-09-01 by bigtreetech

thisiskeithb, 2020-09-07:
The error message was removed from BigTreeTech's firmware, but with https://github.com/MarlinFirmware/Marlin/pull/19279 and https://github.com/MarlinFirmware/Marlin/pull/19281, `EMERGENCY_PARSER` is now fully supported on `STM32F1` (SKR Mini 1.1, SKR Mini E3 DIP 1.0, SKR Mini E3 1.0, SKR Mini E3 1.2, SKR Mini E3 2.0).

rudihorn, 2020-10-07:
Has this issue actually been solved or was the issue just closed?

thisiskeithb, 2020-10-07:
See my last comment.

rudihorn, 2020-10-07:
This issue was never about what Marlin supports, this issue was about the screen firmware supports. I've tried this with a Marlin that supports emergency parsing, but that is useless if the LCD cannot send emergency commands because it is waiting for an `ok` response.

guruathwal, 2020-10-08:
@rudihorn that issue is also solved now.

rudihorn, 2020-10-08:
Okay for the record this was refiled as issue #956 and then solved by #960. Though it would be nice if the gcode screen also supported sending emergency commands immediately.