bigtreetech/BIGTREETECH-TouchScreenFirmware #723: [BUG] (Errors printing from Octoprint, Pronterface on MKS TFT28)

digant73, 2020-05-27:
### Description

Connecting Pronterface or Octoprint to the USB port of the printer and printing any gcode file provides very often beeps and error messages displayed on both the TFT display and Pronterface/Octoprint commands window.
If I print a file from the built-in Pronterface on Simplify3D I have almost no errors (just 2 or 3 during a print).
These errors are randomly produced (even printing the same file) and most of the time is:

Error:Line Number is not Last Line Number+1, Last Line: **xxxx**

where xxxx is a line number (e.g. 47, 2039 etc...)

I also read on Internet that many people had this kind of problem on many different printers (e.g. Prusa) and that it was due to the Printer FW (e.g. on Prusa) or USB connection timeout etc...
But in my case the problem is not present at all on the stock TFT FW of my Artillery Sidewinder X1.

I would like to know if that problem is present also on BTT TFT or it is present only on MKS TFT possibly due to no more support on this BTT repo. At least I will probably decide to revert to stock MKS TFT

### Steps to reproduce

1. Connect Pronterface/Octoprint to the printer via USB cable
2. Load and print a gcode file from Pronterface/Octoprint
3. A lot of beeps and errors will be prompted on both the TFT display and Pronterface/Octoprint commands window

**Expected behavior**
No errors should be dispayed

**Actual behavior**
A lot of errors are displayed along all the print

### Hardware Variant
pin_MKS_TFT32_V1_4

### TFT Firmware Version & Main Board Firmware details
TFT 0.26.1 (current main branch), Marlin 2.0.5.1

### Additional Information
Attached a pictures of the error prompted on the TFT display

[image]

paul-1, 2020-05-27:
I have given up on trying to use TFT mode on my system.   I have the SKR Mini e3 v1.3 and the 3.5TFT v3.0.    

The system works perfectly in LCD emulation mode with Octoprint on the USB port.    I did try to connect Octoprint directly to the TFT's UART port, which it does connect, but there is too much lag in communication.

digant73, 2020-05-27:
Thanks for the check. So the problem should be in the communication between the main board and the TFT.
Waiting from some developers an hint on which source files I should check to make changes in order to fix the problem

odaruc, 2020-05-30:
There is a setting in config.ini that might be of interest. I haven't tried it, but it seems to imply that maybe this is a conflict between two different devices talking to Marlin at once. By connecting through the screen, it becomes possible to keep all commands in sequence perhaps? Thus solving this problem by acting as 'man in the middle':

#### Keep Serial Always ON
# Keep UART (Serial communication) alive in Marlin Mode
# Allow seamless OctoPrint UART connection to the TFT's UART/serial expansion port no matter which mode the TFT is in.
# Options: [0: Disabled, 1: Enabled]
serial_always_on:0

hfog, 2020-05-30:
I coded that. What it does is enable code that polls the UART on the TFT and passes messages through to the printer no matter which mode you are in (TFT or emulation). Works best if you boost the baud rate as high as you can. To use it you need to connect your Octoprint serial connection to one of the UART ports on the TFT. Be aware that you are introducing the TFT in the middle of the communication line so you may see some slight delay between messages due to the TFT having to pass messages back and forth to the printer for you. That can result in some print artifacts if your GCODE contains lots of very short/quick instructions. (There are slicer settings that can improve that.) But at least only one device will be directly connected to your printer this way so you won't get serial port contention. You also will get more complete info on the TFT display because all communication is passing through the TFT this way.

odaruc, 2020-05-30:
Any idea if Marlin is intended to talk to two uarts simultaneously? (provided one of the callers is issuing readonly gcode) -- in which case the TFT takes a role more like a traditional screen but over a serial connection.

Otherwise connecting through the screen seems like a much more reliable idea, and I wonder if the communication can be sped up a lot. Can you link to a commit? I am curious if the implementation uses interrupts and forwards the characters as they are sent/received, or do you buffer until you have a whole command?

rdpeake, 2020-05-31:
@digant73 spotted your issue as i was researching before adding this to my printer - so far as i can tell, the 3 commands from https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/Menu/menu.c#L407 are what deals with 'sending commands to printer', 'parsing and scheduling echo's' from printer and 'getting commands to forward' from connected serial devices like octopi. 
it is possible the 'parseACK' command is inefficient with all the calls to 'ackseen' but no idea if that is the actual issue, or if it's to do with how much else is being done between those 3 commands, as i've not connected it to my printer yet, nor have any debugging stuff set up.

odaruc, 2020-05-31:
The issue seems to stem from Marlin firmware having one global value for last line number. This detects skipped lines while printing, like if the host resets mid print or a command is otherwise lost.

I would argue that line number checking is only needed for the device that is actually printing. However it appears the TFT is sending line numbers by default for all commands, and these are in conflict with the host doing the printing.

One approach is for each host to cooperate who is in charge of line numbers. Marlin does not enforce checks for gcode commands received without a line number, so it would be safe for the screen to issue monitoring commands (or z baby step) while printing. However I cannot say whether Repetier or Octoprint would also abide by this rule.

Another approach is to predetermine which host will be in charge of line numbers, and only print from this device. My proposed solution is for Marlin to ignore line numbers (and M110 commands) coming from a specific usart. If you are using the screen only for monitoring and control purposes (but not printing), this may be perfectly acceptable.

In either case, certain commands sent from the screen could definitely interfere with an active print job there is only one gcode queue. Buttons like movement and homing should not be pressed. Ideally, most buttons on the screen would be disabled until printing is finished.

I'm presently testing with repetier server connected to USB and the TFT35 connected to the SKR Mini E3 TFT header.

Here's a pull request submitted to the marlin 2.0 bugfix branch with my proposed fix.
https://github.com/MarlinFirmware/Marlin/pull/18165/commits/e4aa62ed4d388918a8c167968c0553b6b6794ecd

odaruc, 2020-06-02:
PR has been revised to actually do line number checking per port. Appears to have solved the issue for me.

digant73, 2020-06-03:
@odaruc, @hfog
I monitored the commands sent and received by the TFT (setting terminal_ack to 1 on config.ini) on the Terminal menu. According to the menu currently displayed, the TFT periodically sends some gcodes (e.g. M105, M220, M221 are sent in the main menu) that will be replied by Marlin. However, the TFT also receives the "echo OK" from Marlin related to the gcodes sent by Pronterface. Both the reply and echos are processed by the parseACK() function in the TFT. Sometime the gcode sent by TFT seems colliding (on Marlin side) with the gcode sent by Pronterface. In this case, Marlin will reply to both TFT and Pronterface with an error. Sometime the error is the "Line + 1...", sometime it is reported a corrupted gcode (e.g. "èpM105@" etc...). The unique commands queue on Marlin explains to me the "Line + 1..." error but not the corrupted gcode. Is there any explanation also for that kind of error?

odaruc, 2020-06-03:
@digant73 I did some digging..  Is the controller a Makerbase MKS SGEN L board? I'm unfamiliar with the Sidewinder.

I checked the schematics for that board and it looks like AUX-1 is dedicated to its own usart separate from USB. Not sure how these could be interfering or bonded. The pins are on physically opposite sides of the CPU.
https://github.com/makerbase-mks/SGEN_L/blob/master/Hardware/MKS%20SGEN_L%20V1.0_001/MKS%20SGEN_L%20V1.0_001%20SCH.pdf

Does the original TFT firmware have persistent information display? That may explain why the original TFT firmware works, lack of possibility of a collision. In that case, I don't think the hardware is to blame, maybe the marlin firmware or configuration.h though.

digant73, 2020-06-03:
> @digant73 I did some digging.. Is the controller a Makerbase MKS SGEN L board? I'm unfamiliar with the Sidewinder.
> 
No, It's MKS GEN L not MKS SGEN L

> I checked the schematics for that board and it looks like AUX-1 is dedicated to its own usart separate from USB. Not sure how these could be interfering or bonded. The pins are on physically [...]
There is a usart (RX, TX pins) connected from MKS GEN L to the TFT.

> Does the original TFT firmware have persistent information display? That may explain why the original TFT firmware works, lack of possibility of a collision. In that case, I don't think the hardware is to blame, maybe the marlin firmware or configuration.h though.

If you mean the original TFT firmware displayed info such as temperature etc...I would say yes it displayed those info.

However, in order to flash the Marlin firmware from it's USB port (the same used by Pronterface) it is required to disconnect the TFT cable (or from BTT firmware press on the discconnect button) because the TFT do not allow flashing the Marlin firmware. This let me think that the TFT<->main board usart and the main board<->USB serial to PC can normally collide due to Marlin handling.

odaruc, 2020-06-03:
AUX-1 on the MKS GEN L appears to be wired to the same pins as USB RX TX.

I have one of these boards currently installed in a printer. It's running repetier firmware and I repurposed a couple of the end stops for the TFT screen so it could be connected simultaneously. Repetier has its own separate bug with serial comm... this is a problem I was trying to solve for quite a while before I gave up and moved on to a new 32 bit board with the trinamic steppers and Marlin 2.0

digant73, 2020-06-05:
> AUX-1 on the MKS GEN L appears to be wired to the same pins as USB RX TX.
>
> I have one of these boards currently installed in a printer. It's running repetier firmware and I repurposed a couple of [...]

Ok, so the serial bus is shared with USB. Does it mean corruption of data is possible if both the Pronterface and TFT try to use the bus simultaneously?

odaruc, 2020-06-05:
Yes it will interfere. Found a related issue suggesting the same thing I am about to suggest: https://github.com/MarlinFirmware/Marlin/issues/13608

Here is how to get around that. I assume you are not using the max endstops at all.
1. It will be necessary to plug the y-min endstop into z-max for this mod.
2. Reconfigure z-max to act as y-min (Pin D19 becomes y-min)
3. Connect the TFT screen to y-min/y-max pins D14/D15 (TX3/RX3 pins of serial port 3)

From the pin mapping: https://www.arduino.cc/en/Hacking/PinMapping2560

I didn't test this with Marlin, but the configuration.h would probably look like:
```
#define Z_MAX_PIN 2 // assign to same pin as x-max to free up z-max
#define Y_MAX_PIN 2 // assign to same pin as x-max to free up y-max
#define Y_MIN_PIN 19 // y-min becomes z-max now
#define SERIAL_PORT 0 // main serial port will be TX0/RX0
#define SERIAL_PORT_2 3 // extra serial port will be TX3/RX3
```

[image]

closed 2020-08-16 by digant73