bigtreetech/BIGTREETECH-TouchScreenFirmware #724: Update with config.ini

Kyosuke33, 2020-05-28:
Hello, i try to update my tft24 with the config.mini file, but nothing happen.
I open config.ini with Visual Studio Code, i change what i need ( printer size and probe points), i copy the config.ini on my SD carte formated with FAT32, put the SD card in the printer, plu in and nothing happen, i got the statut screen but the config doesn't change.

Any idea ? Thank you.

Tygrys-1, 2020-05-29:
What software version on TFT?
Which slot you put SD card in?

Kyosuke33, 2020-05-29:
TFT24 1.1 with 26.1 firmware.
I put the SD card on the SD card port of the screen.
I just see one other guy with this problem two message below mine.

Tygrys-1, 2020-05-29:
Initial release of 26.1 (the one you can find in the "Copy to SD folder") does not have ability to use config.ini, so you must build your own firmware from the master branch.

Kyosuke33, 2020-05-29:
OK :'(

oldman4U, 2020-06-01:
Hi Kyosuke33.

Could you please close this ticket as it is not needed any more.

Thank you

closed 2020-06-01 by bigtreetech