bigtreetech/BIGTREETECH-TouchScreenFirmware #732: [FR] (Simplified Status Screen while Printing)

gordo3di, 2020-05-30:
So far we love the Unified Menu status screen while printing, however would it be possible to implement a very simple screen that appears after "x" seconds. Basically just showing name of file, percentage complete and maybe time all with very big text on a black screen. This way you can see basic info from far away. This would be super useful for people with multiple printers. Also the persistent hotend and bed temp could still be on top.

Things like FR, Fan, Temp, Layer height, Pause, Babystep, More and Stop aren't really important while printing unless something goes wrong.

Then if you touch the screen it would immediately go back to the normal status screen and wait for the new countdown...

Just a thought. I'd love to program it however I think you would spend more time fixing my code than anything... How can I help?

[image]

oldman4U, 2020-06-15:
Hello gordo3di.

I believe this would be a good idea, but based on the current font used for the UI this could look ugly: But you would like to see the attributes shown in your screenshot, where the Progress is available twice and therefore is redundant.
[image]

gordo3di, 2020-06-15:
That makes sense. We were just looking for something we could easily see from far away and take away stuff that doesn't matter for general printing. I take it scaling fonts doesn't really work?

oldman4U, 2020-06-15:
What if it would be possible to improve the readability of the whole UI?

Such a simple status screen, how would you activate it? Automatically after xsec the print started or manually by pressing a button on the interface?

oldman4U, 2020-06-15:
Sorry....

As far as i know you need an additional, bigger font. And space is limited - as usual.

gordo3di, 2020-06-17:
> What if it would be possible to improve the readability of the whole UI?
> 
> Such a simple status screen, how would you activate it? Automatically after xsec the print started or manually by pressing a button on the interface?

Our thought was to have it appear after x seconds kind of like a screen saver. Currently our printing screen looks like the attached image.

[image]

It works but sometimes it hard to see the progress at a fast glance especially if you have many printers running.

Thanks for the input

oldman4U, 2020-06-17:
I completely understand the function and readability is an issue with the current screen. Hope we will see such a WhilePrinting screen in firmware soon.

oldman4U, 2020-10-20:
Hello gordo3di.

Could you please be so kind and close this ticket. Your feature request is covered in the pinned Feature Request ticket #1177 

Thank you

gordo3di, 2020-10-20:
@oldman4U whoops forgot it was open. I made some headway on it but not enough to be useful.

closed 2020-10-20 by gordo3di

oldman4U, 2020-10-20:
My design for your requested feature. Hope you like it😁

oldman4U, 2020-11-05:
Just saw that there was no attachment.... Sorry

[Gordo.pdf](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5497916/Gordo.pdf)

gordo3di, 2020-11-05:
Ooohhhh!!! I like it.

oldman4U, 2020-11-05:
[Gordo.pdf](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5497961/Gordo.pdf)

And this is with the correct lower bar...

oldman4U, 2020-11-05:
It will come, but before there will be a VERY VERY cool Print Confirmation screen coming.

gordo3di, 2020-11-06:
Awesome! I'm even more excited now... I'll try to figure out github
desktop. I want to help contribute too.

oldman4U, 2020-11-06:
Hehe👍🏻