bigtreetech/BIGTREETECH-TouchScreenFirmware #736: [FR] (support HOST_ACTION_COMMANDS for Marlin)

dp250f, 2020-05-31:
In order to use Host mode instead of Marlin mode, HOST_ACTION_COMMANDS support is needed to accomplish filament change and other actions which require prompting.

If TouchScreenFirmware could support HOST_ACTION_COMMANDS, it would be much more functional.

Currently if you attempt an M600 (filament change) it unloads, then gets stuck in an echo: loop and never loads or continues printing. If you switch to Marlin mode at this point, you can see it's waiting for you to press the button to tell it you've loaded the new filament.

Thanks.

dp250f, 2020-06-03:
This would also allow filament runout functionality without running filament runout wires to the screen.

ETE-Design, 2020-08-06:
@dp250f  Seems to be possible with the right code.... https://www.youtube.com/watch?v=HJpfO8XifqI

radek8, 2020-08-20:
https://github.com/PrintEngineering/MKSTFT28_Clone_Artillery-Sidewinder/tree/master/TFT/src/User

bigtreetech, 2020-09-01:
Hello HOST_ACTION_COMMANDS  should be worked after https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/984

closed 2020-09-01 by bigtreetech