bigtreetech/BIGTREETECH-TouchScreenFirmware #742: Where i can find previous firmware.

Kyosuke33, 2020-06-01:
Hello, i just upgrade the firmware of my TFT35 from 26.1 to 26.x.
That corrects the problem of the wrong name icons and the config.ini problem to update new configuration.
But now when i turn on the printer, i have an error "Fade height OFF".
So i search for the 26.1 firmware to open it with VSC and compare it with the 26.X and find the error.
There is a place where the old firmware are saved ?
Thank you.

Tygrys-1, 2020-06-01:
There is a branch named release_xx_26.1. you will find the files from 26.1, build about month ago.

You may also click on "commits" to see the history of changes and trace to the release 26.1 point.

Also, selecting a file from the repository and clicking a "History" button will show changes for that file.

When browsing commits you may use the "<>" button to view the repository at that stage.

Kyosuke33, 2020-06-01:
Great Thank you !

closed 2020-06-01 by Kyosuke33