bigtreetech/BIGTREETECH-TouchScreenFirmware #743: Illegal flash app after inserting a SD card with config.ini on it.

draude71, 2020-06-01:
I had this problem before but i just installed the new BIGTREE_TFT35_V3.0.26.x.bin firmware update and tried it again.

At first i placed the BIGTREE_TFT35_V3.0.26.x.bin file, TFT35 folder and config.ini file on the sd-card but this resulted in a black screen with the message: Illegal Flash App.  Power down/reset didn't work, the message keeps appearing.

I cleared the sd-card, put the BIGTREE_TFT35_V3.0.26.x.bin file and TFT35 folder on it and now the flashing was successful.

I put the config.ini on the sd-card, insert it in the screen, did a reset and again, the same message: Illegal flash app.

I'm a doing something wrong?

KevMags, 2020-06-01:
Same here with TFT 24.  It happens with V25 and V26.  At least with V25 it will config when flashing.

sangeriz, 2020-06-13:
Same for me, all the time when you turn on the printer show illegal app and its imposible work, is the second board for me, anyone with the solution?

KevMags, 2020-06-13:
I used the V25 firmware and config.ini ONLY when installing the firmware.  If you have config.ini on the SD card after flashing it gives "illegal flash".   This is on TFT24.  I only use the Marlin emulation,  the touch firmware is unusable.

sangeriz, 2020-06-18:
I Get out from "illegal flash" my TFT's 35 v3.0, just download from GitHub, all folders to compile, check the files with visual studio code, compile  and with BIGTREE_TFT35_V3.0.26.x.bin working again.

zozo6015, 2020-06-21:
it happened to me when I had other things on the sd card. Once I deleted everything and I only kept the firmware it did work.

oldman4U, 2020-07-23:
Hello @draude71 

Could you please close this ticket in case you do not need it anymore. Thank you

closed 2020-09-01 by bigtreetech