bigtreetech/BIGTREETECH-TouchScreenFirmware #745: [BUG] (short description)

Talha909, 2020-06-01:
I have SKR 1.4 Turbo with TFT 24 Touch Screen. I have updated the new software BIGTREE_TFT24_V1.1.26.x. I put the memeory card in the tft memory card slot after that. Update begin. After update touch screen alignment comes.  Accidentally i switch off the printer on that screen and did not perform the touch screen alignment. Now screen reboot and classics menu comes in-front of me.  Now the touch is not not working and printer not attached errors comes.  I press the nob and try to go marlin screen. But i put the on marlin screen and press the switch button but it do noting. Its on the marlin and classis menu screen. I cant able to to do nothing. Any suggestions please.

oldman4U, 2020-06-02:
Hi.

Have you tried to flash the firmware again?

Talha909, 2020-06-02:
I will check it todsay

radek8, 2020-06-03:
Insert the reset.txt file on the SD card and restart the display. The display touch calibration is performed.

Talha909, 2020-06-03:
OK THANKS

oldman4U, 2020-06-08:
Hi. Any news?

Talha909, 2020-06-08:
i go back to the old firmware.

oldman4U, 2020-06-08:
So it works with an older version of the firmware?

Have you tried to use MASTER Branch?

oldman4U, 2020-06-15:
Hello Talha909.

It would help us and also other users like you, if unneeded tickets would be closed by the owner.

@bigtreetech Please close this unneeded ticket. Thank you

closed 2020-06-15 by Talha909

dieund, 2021-05-06:
thanks. it work.