bigtreetech/BIGTREETECH-TouchScreenFirmware #747: [Q] Support for other simulator protocols?

rdpeake, 2020-06-02:
**Your question or details you need. Please describe.**
Is it possible, (or has a fork attempted to) add an option for alternative LCD signals, like parallel (HD44780U) I bought this with the intention of using it with a prusa mk3, however they have removed SPI LCD's from their source and only use a parallel interface.

If there is not any existing setup for such an interface, what would be the appropriate place to add such an interface without overloading the code base (assuming there would be any intention of merging it back in)?

**specify the device variant related to your question.**
i use the BTT tft 3.5 v3.0

bigtreetech, 2020-06-03:
Yes, it's possible, because our screen has reserved parallel IO, but these are GENERAL io. so I don't know whether the io speed can meet the requirements of 6800 interface before test

rdpeake, 2020-06-03:
Given the prusa is driving the IO from an einsy rambo, i would assume the timing input of general IO is likely to be sufficient, though it may not be fast enough to match full spec speeds of the hardware being emulated?

bigtreetech, 2020-06-04:
yes, It needs debugging and testing

rdpeake, 2020-06-06:
i don't mean to sound 'pushy' just unsure as to what is implied by your response.
Is this something you plan to work on, or should i be asking for guidance on where to start attempting to add this sort of thing myself?

bigtreetech, 2020-06-09:
Hello， I plan to do it in the future not now. It could be at the end of this month or next month, ^_^

bigtreetech, 2020-08-01:
Hello，This function has been implemented in  #864

rdpeake, 2020-08-03:
Is there an update binary available to test this (or should i ask in the pull request)?

bigtreetech, 2020-08-19:
Hello， This feature is already included in the precompiled firmware, you need only set Marlin mode from 12864 to 2004 in config.ini and update it.

closed 2020-08-19 by bigtreetech