bigtreetech/BIGTREETECH-TouchScreenFirmware #748: [BUG] Cannot upload the complete system TFT24 new

mozarttelles, 2020-06-03:
### Description
TFT not working, marlin is working with reduced size screen.  It is already tiny 2.4", now it is smaller.

### Steps to reproduce

Formated a 16Gb disk, installed the directories TFT24 with subfolders BMP and FONT, reset the screen and updates the first firmware, I believe it is the MARLIN, because it is working. then goes to upload the touch screen and stops at 3-4% waited hours and repeated multiple times.

**Expected behavior**
Push and hold the bottom to transfer to TFT, it shows the TFT but not any image. it is a mess.

**Actual behavior**

### Hardware Variant
GEN L board 1.4 

### TFT Firmware Version & Main Board Firmware details
TFT 2.4 with marlin

### Additional Information

radek8, 2020-06-05:
You have copied the file to the SD card
BIGTREE_TFT24_V1.1.26.x.bin
and a folder
TFT24
Did he insert the card into the slot on the TFT 24 display and restart the display?

mozarttelles, 2020-06-05:
Hello, thank you for returning my issue.  

 

Yes I did.  It show updating but get stuck in the fonts.  It does not pass the font part. 

 

I will take a picture or video later, when I finish a print.

 

Thanks.

oldman4U, 2020-06-08:
Hello mozarttelles.

Have you been able to fix the reported issue?

If not, have you tried this?

To reset the TFT's touch screen calibration, create a blank file named reset.txt and place in root folder of the sd card. Insert the SD card into the TFT's SD card reader and power cycle your printer to start the reset process.

Please let us know

mozarttelles, 2020-06-08:
Tft 24 had to be returned, the issue was not fixable. I bought a tft35 v3
and worked fine, updated no problem.

oldman4U, 2020-06-09:
Sounds like a hardware issue then....

Good to hear you have a working solution now. Could you please close this ticket by clicking the Close button right besides the Comment button.

Thank you and happy printing

mozarttelles, 2020-06-09:
Thank you for following up on this issue. The issue was hardware for sure.  I had changed to a new board TFT35 V3 and worked flawlessly.

oldman4U, 2020-06-09:
And you got not only a bigger screen but also cool features like dimming, faster speed, and so on;-)

Could you please be so kind and close this ticket by clicking on the Close button right besides the Comment button.

Thank you

mozarttelles, 2020-06-09:
Sorry, I forgot.  Yes that is true. 

 

I love the bigger and better screen.

closed 2020-06-09 by mozarttelles