bigtreetech/BIGTREETECH-TouchScreenFirmware #754: TFT35 Touch not working after update

MontyFlange, 2020-06-04:
I can use the screen now. is there any way to force a calibration for the screen after an update ?

oldman4U, 2020-06-04:
Hi.

As far as i know there is a calibration function in the user interface itself under Screen.

radek8, 2020-06-04:
Insert the reset.txt file on the SD card and restart the display. The display touch calibration is performed.

MontyFlange, 2020-06-04:
Where does one find this 'reset.txt file then ? It's not in the bundle I downloaded

oldman4U, 2020-06-04:
> Which I can't use because the screen doesn't work

And does not work means in detail?

Black screen
Does not react when touching
Executes the wrong function
Only touchscreen but no marlin
Or
Or
Or
?

MontyFlange, 2020-06-04:
> And does not work means in detail? Black screen Does not react when touching Executes the wrong function Only touchscreen but no marlin Or Or Or ? MontyFlange <notifications@github.com> schrieb am Do. 4. Juni 2020 um 18:37:
> […](#)

No response to touch

MontyFlange, 2020-06-04:
> > Where does one find this 'reset.txt file then ? It's not in the bundle I downloaded
> 
> Create an empty file with this name:
> reset.txt

LOL that simple OK :)

MontyFlange, 2020-06-04:
OK RESET TXT worked :) Great Thanks for all the help

oldman4U, 2020-06-04:
Hi MontyFlange.

This is great. So everything works now? I would also recommend to use Master Branch instead of 26.x firmware version. Many bug fixes and configuration can be changed using config.ini file.

Please let us know

MontyFlange, 2020-06-04:
Master branch ? All this gargonisum is new to me :)

oldman4U, 2020-06-04:
I totally understand what you mean!

Most important - EVERYTHING works now?

If you have not been working with Visual Studio Code before or you are not interested to spend more time by tinkering around with software - don’t start with it and forget my last reply.

But besides the official firmware, there is always firmware available which you can download and “compile” by yourself. This requires a computer, free software and some time to learn how to do it. The advantage is, that you can make changes to your software, not available I the user interface you are using today.

In your case there is the official “ready to use” firmware and a firmware which is under development. This under development version for the TFTs is called Master. Attached a screenshot which shows what I mean.

MontyFlange, 2020-06-04:
I have played with PIO to make Marlin. Problem started when the version of Marlin given out by BIQU on the GitHub page was to old to work in the new PIO and wouldn't compile, which started this 'update' trail :)  Got Marlin working, then their 'tech' guy said update TFT, So came here and did that following their instructions , then it all went belly up :)

MontyFlange, 2020-06-04:
But I do think that the reset.txt file and instructions should be included in the package :)   6 hours of trying to get it working, then 30 seconds with that info.

oldman4U, 2020-06-04:
The TFT software is maybe 1/50th compared to the software for your board, maybe you want to give “Master” a try. Otherwise enjoy your TFT and happy printing;-)

I agree that documentation could be better and that a proper manual would avoid maybe half of the tickets here.

Please close this ticket if you do not need it any more and start a new one if needed

Thank you

MontyFlange, 2020-06-04:
Will do. I think the ini file has all the changes I'm willing to risk for now :) Thanks

radek8, 2020-06-04:
Information about the reset.txt file is on the Welcome screen. Just read to the end :-)

[image]

oldman4U, 2020-06-04:
Hi Radek.

This is very cool, thank you. I never needed it, but good to know;-)

Thank you

MontyFlange, 2020-06-05:
> Information about the reset.txt file is on the Welcome screen. Just read to the end :-)
> 
> [image]

LOL yer saw that later Do'H :)

closed 2020-06-05 by MontyFlange