bigtreetech/BIGTREETECH-TouchScreenFirmware #756: Auto shutdown does not work

Moreman73, 2020-06-04:
Hello, after updating the TFT 35 V3.0 E3 firmware from version 26 to 26.x, the auto shutdown function stopped working, the screen freezes when it is automatically turned off, if the power is turned off manually through the menu, then everything turns off normally

### Hardware Variant
<TFT35 V3.0 E3>
### TFT Firmware Version & Main Board Firmware details
<TFT Firmware Version 26.x Main Board Firmware Marlin-bugfix-2.0.x
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4728942/Configuration.zip)
>

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

Moreman73, 2020-06-23:
An error is present when using the Russian language; if you switch the display to English, then auto power off works fine

bigtreetech, 2020-07-04:
Hello,  can you test the lastest version firmware?  I just tested it with the latest firmware in Russian settings and found no exceptions

Moreman73, 2020-07-09:
I have tested it, now everything works fine, thanks

closed 2020-08-19 by bigtreetech