bigtreetech/BIGTREETECH-TouchScreenFirmware #757: When the encoder adjusts the fan speed, the fan speed and print speed are changed together

Moreman73, 2020-06-04:
Hello, if you go into the fan speed settings of the model blowing fan and start adjusting its speed with the encoder, this causes a change in print speed along with a change in the fan speed, everything works fine with the touch buttons.
<TFT35 V3.0 E3>
<TFT Firmware Version 26.x Main Board Firmware Marlin-bugfix-2.0.x

radek8, 2020-06-04:
Unfortunately, it's not a mistake, it's a feature
read here
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/680

try pressing 1 briefly before using the encoder.
This will take you to a menu in the marlin menu where there is no change in speed.
But more presses can be a problem :-)

Moreman73, 2020-06-04:
> Unfortunately, it's not a mistake, it's a feature

The fix that you suggested does not work? "Fixed encoder behavior in touch mode. #718"

radek8, 2020-06-04:
It's not working, so I closed it

AnHardt, 2020-06-04:
I suppose the contacts of the encoder, the pins on the ext port to the mainboard where Marlin runs and the inputs of the displays processor are hard wired together. In that case there is no satisfying solution. The encoder signals will always be received by the display and the mainboard. So either one can use the encoder with the display - then Marlin should be configured to nut use its display. Or Marlin can use the display but the touch screen firmware has to ignore the encoder. Else both of the systems react independently on the encoder and do different things.
A proper hardware solution would be to route the encoder signals only to touchscreens processors input pins and connect the pins to the mainboard only to some outputs of the touchscreens processor. Then the touchscreen firmware had to mirror the inputs to the outputs only when in Marlin mode. An other way, saving some pins, could use some hardware AND-gates to block the signals to Marlin when not in Marlin mode.

radek8, 2020-06-05:
You're saying that right.

But this would be a big HW and SW intervention in the display.

The solution is to make sure that before you turn the encoder, Marlin is not in the home screen, where he changes the print speed.
If it is in the menu, turning the encoder will only move the menu but will not change the speed. you must not press the encoder, otherwise you will start the function you are standing on in Marlin.

Another option is to disconnect the EXP2 connector if I don't want to use the Marlin transmitter. EXP2 transmits encoder movement data to Marlin. If I switch to Marlin, you'll plug in the connector. :-)

AnHardt, 2020-06-05:
You'll never know if Marlin is still in Menu-mode or if already returned to status-screen via timeout. This will never work reliably.

Adding a g-code to Marlin to enable/disable the display-panel-inputs? Kind of 'keylock'?

radek8, 2020-06-05:
When I disable the option
#define LCD_TIMEOUT_TO_STATUS
does the return to the status screen turn off?

radek8, 2020-06-05:
This option will only increase the time to return to the status screen.
after disabling it, the default value of 15s is used
A change would have to be made to the Marlin code

radek8, 2020-06-05:
Option -1 turns off the automatic return of the Marlin setting screen
#define LCD_TIMEOUT_TO_STATUS -1
If I set Marlin like this, just 1 press of the encoder, the next turn of the encoder will no longer change the print speed in Marlin.
This does not solve the problem, but allows you to use the encoder in touch mode without changing the print speed.

radek8, 2020-06-05:
If pressing the encoder were emulated when the printer was connected, further rotation of the encoder would no longer affect the printing speed.

Currently, the first time you switch from the touch screen to Marlin, one turn and one press of the encoder is sent, but I don't know why.

[image]

Tygrys-1, 2020-06-05:
Maybe to force the Marlin to refresh the display?
However, this does not work.

And why no longer can select modes using the knob?

kind3r, 2020-07-01:
I was wondering why the speed display went crazy when tryint to adjust, and also why the speed changed by itself all the time.

In Marlin there is an option that should disable feedrate multiplier on the status screen in [Configuration_adv.h](https://github.com/MarlinFirmware/Marlin/blob/0d3a5de4bac158637712dae969cc9808f47f83a2/Marlin/Configuration_adv.h#L946 ) called `ULTIPANEL_FEEDMULTIPLY`. Haven't tested yet but I will in a couple of hours. While not perfect it should solve most of the issues (unless you really need to change the feedrate from the Marlin status screen), just make sure you are on the status screen when switching to TFT mode.

kind3r, 2020-07-06:
Before the https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/41a11faf2384957114bb23421895fa861b6458d4 adjusting fan speed, speed and flow using the knob in the BTT default firmware worked, now if you turn the knob fast it glitches. I was using @guruathwal 's branch when I noticed this, switched to the BTT and it worked fine. Now the glitch has propagated in the main fw. I tried to investigate the issue but can't figure it out. It could be related to the new larger fonts cause everything else seems to work the same and that is the only obvious change (except for the code split).

kind3r, 2020-07-06:
Made a short video about the issue: https://youtu.be/u0Gz-jvzVVs

The changes are also sent to the printer and the glitch persists until reboot.

oldman4U, 2020-09-05:
@Moreman73 

This issue is now described in the pinned ticket #1047 

Please close this ticket

oldman4U, 2020-09-11:
@Moreman73 

Close this ticket

closed 2020-09-12 by Moreman73