bigtreetech/BIGTREETECH-TouchScreenFirmware #758: [BUG] No resume after change filament (M600)

VijitCoder, 2020-06-04:
### Description

I used Cura 4.x. 

Sliced a model, set M600 command for changing filament. But the printing is never resume. On the screen constantly shows _"Printer Paused. OK to continue"_. By pressing Ok this message shows again.

Also I noticed that the nozzle become colder, which is strange. Then the temperature upped to the specified value but the message toggle and toggle again.

What else. Before begin to print my model, I made a test with this feature. I took a cylinder d20, sliced it and added Change Filament command. I reduced temperature of the nozzle and the bed cause I made this print without filament. And it works as expected. Paused and resumed.

The problem model from [here](https://www.thingiverse.com/thing:4247272). _ImpossiTableTop.stl_ Change filament at layer 4. Layer height - 0.25.

Both gcodes will be attached.

### Steps to reproduce

Well, you can use my gcode, printing an hour and half. Then see what will happen on the filament change.

**Expected behavior**

Printing must resume by pressing "Ok" in the popup.

**Actual behavior**

No printing resume.

### Hardware Variant
Bigtreetech TFT35-E3 V3.0
### TFT Firmware Version & Main Board Firmware details
Screen - 26.x
Marlin - 2.x

### Additional Information

[files.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4729203/files.zip)

marsolek84, 2020-06-05:
I too have this issue.  There are two band-aid fixes I have found that work.  I have printed it in the "Marlin emulator" and that does bring up the pop-up to resume.  The other is for touch mode.  When it gets to the filament change, you click ok, and change filament.  If you push the encoder knob in, it will purge more filament and keep doing so until you resume.  After you have purged and gotten clean filament, rotate the encoder knob clockwise one notch (bump) and push the encoder button in.

No pop-ups appear on the screen, but it will resume print.  Sure seems like a bug to me, but this works.  It isn't ideal by any means and would like to know a fix too.

Give it a try and let me know if it works for you.

HappyCodingRobot, 2020-06-05:
Marlin 2.0.x requires an LCD connected for the advanced pause features. (see Configuration_adv.h)
Afaik this cannot work with the serial connected TFT and actual Marlin versions.

VijitCoder, 2020-06-05:
@marsolek84 Yeah, I'm pretty sure it will work in emulation mode. But the other way that you've described.. I donno, it's like a wandering in the dark :)

VijitCoder, 2020-06-05:
> Afaik this cannot work with the serial connected TFT and actual Marlin versions.
Well, half of the command M600 definitely works on RS-232. TFT sent the command, and printer reacted properly: paused and parking.

VijitCoder, 2020-06-05:
I suppose, the screen firmware sends a wrong command for resume. But it just a guess, I not dug in the code about that problem.

marsolek84, 2020-06-05:
Definitely like wandering in the dark!  I can confirm though that it does work, there just are no prompts on the screen.  I figured it out by chance, trying it in the emulator mode first, then "remembering" my steps and doing so in touch mode.

I am running the SKR Mini e3 and a TFT35 by the way.

Let me know what other info I can provide, as well as if a fix is available.  I am not a coder by any means...a noob for sure.

Tygrys-1, 2020-06-05:
As the knob is connected to the Marlin directly, when the printer displays it's pause screen with a purge/enough choice (and you do not see it beeing in touch mode), you can blindly select the right choice using the knob. Also, if there was no reaction for some time the heater will turn off, and you must blindly restore the change procedure using the knob, then wait for heater to heat up and filament feed in.

VijitCoder, 2020-06-05:
> if there was no reaction for some time the heater will turn off

I can confirm that. But why it's happens? There is no prerequisites to do so. Another strange behavior of the Screen firmware?

> and you must blindly restore the change procedure using the knob

I think here is exists a chance to switch to simulation mode and made all operations in "a straight view". But I don't want even check this assumption cause the whole this situation - is a bug due to the raw firmware, IMO. I found more bugs and strange UX behavior. This firmware even not can be called beta.

marsolek84, 2020-06-05:
Ok, I understand now, that the knob is connected differently than the TFT screen.  I read the post about "adding emulated M601" from @HappyCodingRobot, does that mean there will not be a fix for this via the TFT screen?  I enjoy printing multiple colors and would prefer not to go at it blindly or in emulated mode.  I am glad I figured out that I can still do it in TFT mode, however I think it should be fixed.

Tygrys-1, 2020-06-05:
This is the Marlin limitation. I'm afraid it cannot be done on TFT by itself. Maybe try switching temporary to Marlin when this happens, and then return to touch mode? (not sure this will work)

VijitCoder, 2020-06-05:
FYI: switching to Marlin emulation fails in 50% from the working TouchScreen mode. It gives just a blank screen with a title above. Reset helps to enter to emulation, but not in during the printing of course. _This is another bug._

So, the only way is to apriori use the Marlin mode for multicolor printing.

VijitCoder, 2020-06-05:
What about this part from the bug description:

> Before begin to print my model, I made a test with this feature. I took a cylinder d20, sliced it and added Change Filament command. I reduced temperature of the nozzle and the bed cause I made this print without filament. And it works as expected. Paused and resumed.

marsolek84, 2020-06-05:
I will give it a try, let me see if I have this correct.  I am to

Open a project, slice it, add my filament change, reduce the nozzle and bed temp, slice, save, and run?

I'll give that a shot right now.

Tygrys-1, 2020-06-05:
> 
> FYI: switching to Marlin emulation fails in 50% from the working TouchScreen mode. It gives just a blank screen with a title at the top. Reset helps to enter to emulation, but not in during the printing of course. _This is another bug._
> 
> So, the only way is to apriori use the Marlin mode for multicolor printing.

It works 100% for me. Just after the switch rotate the knob a little. Marlin needs some reason ro repaint the display, if nothing has been changed there is no reason to redraw.

marsolek84, 2020-06-05:
In TFT mode, still no prompts to resume, it will prompt at the START of the filament change, but no direction what to do next.  When I click OK on the start filament change prompt, it'll unload when I click ok.  From there I reload filament and push knob.  Filament loads in, push knob again and it'll keep purging, rotate knob one click, push knob to confirm, and it'll restart the print (with new color).

Tygrys-1, 2020-06-05:
> TFT worked out as expected. Why, if this is a core Marlin problem and can't be solved?

TFT is a terminal to the Marlin. If it prints from the TFT, it reads the file from the SD or USB and code by code send this to the printer. Gcode is a one - way interface. You send the command, you get the reply. Marlin does not ask the sender "Press Y for Yes" or "Press any key to continue", in fact, there are no keys on gcode :-). To execute the interactive command Marlin uses it's own display to ask for choice or continuation. It has no way to send a request to the SD card or any other gcode source like TFT to get the response. That way it is hard for TFT to react to the M600, as the Marlin does not wait for any gcode at this time that the TFT can send to it. It waits for the knob press, and you can choose to purge more or stop purging and continue.

It is not a problem, it is how things are designed.

VijitCoder, 2020-06-05:
> Marlin needs some reason ro repaint the display, if nothing has been changed there is no reason to redraw.

Interesting.. Well, may be it make sense to give a reason for Marlin _automatically_, nop? [M27](https://marlinfw.org/docs/gcode/M027.html) or something like that.

These tiny nuances confuses the user and form a negative opinion about the product as a whole.

marsolek84, 2020-06-05:
I agree. Plus I have very limited exposure to the coding or anything printing for that matter.  I am glad I have a better understanding of how it works now, and that I know I can do it from TFT mode, just not as pretty as I would have expected.

All good though!  Thank you for all the insight.

Tygrys-1, 2020-06-05:
It lacks documentation. But once learned a trick, one can get used to.

M27 is a terminal command and when you send M27, you get the reply on the terminal. Still no reason to update the printer's display, as the displayed information has not been changed.

VijitCoder, 2020-06-05:
> Still no reason to update the printer's display, as the displayed information has not been changed.

Ok. The only reason to update the screen is a knob rotation? No other suitable options?

Tygrys-1, 2020-06-05:
You can send M117. But this will erase the previous message. And this is a "feature" of the Marlin, I do not know if TFT can do anything about it. Also, redrawing the display if there is nothing new to draw is a waste of cycles.  :-)

VijitCoder, 2020-06-05:
I can understand when there is no real reason. But in that case (switching mode) the reason exists.

---

Try to see it from the my view point. I bought a device with a nice promises from the good company. Updated to the last firmware and.. my WTFs per hour has exceeded any possible limits. I even dropped all my deeds and started digging in the sources till the deep night, because I needed to do something with that. This firmware is awful. The screen shouldn't cost that money, the Bigtreetech must send it for free for anyone cause we're all become beta testers and/or maintainers.

And if there is any chance to make the firmware a bit better before buyers run away from the seller of that screen, I believe, it must be done. The easiest way - a complete documentation with the honest list of disadvantages. As for now, I known much more from this thread then from the official sources.

Tygrys-1, 2020-06-05:
Yep, it can be frustrating. I hoped for more bells an whistles from this thing, and I end up using it in Marlin emulation ;-). It is however usefull in certain cases like positioning the nozzle or providing WiFi to the computer. Also it has a handy gcode terminal and big future potential from the hardware. It is open source so can be programed for different things because of potent hardware. So, overall, not tha bad deal for about 25 bucks. And I learn a lot playing with it.
For sure, it looks better than it works :-)

Tygrys-1, 2020-06-05:
> I can understand when there is no real reason. But in that case (switching mode) the reason exists.
> 
Yeah, but Marlin is not aware of the fact, that someone started looking at it's display. ;-)

In touch mode it is the terminal to the printer. In Marlin mode it is a display to the printer. There is kinda loose coupling between TFT and the printer, don't expect the high level of integration. I think, that maybe it can be more integrated, but this will require lot of programming on both sides. The product is quite new, so maybe?

VijitCoder, 2020-06-05:
> I end up using it in Marlin emulation

I'm inclined to think the same :) And wait a couple years more while that open source will be polished.

Tygrys-1, 2020-06-08:
To reveal more of the mistery of blank screen on mode change, it happens when Marlin is in the menu. To redraw the menu you need to move the selection, or press the knob to go back to the status screen, or do whatever is selected in menu you don't see. 
If the Marlin is showing the status screen, it refreshes itself and there is no blank screen on mode change. 

So, to avoid the blank screen: change the mode by touch only, avoiding pushing the knob. If you leave Marlin on menu, it will revert to the status screen after a while. In case you see the blank screen, that means you are in Marlin's menu. You can rotate the knob to refresh it, press the knob and that will do some action like in the menu (mainly return to the status screen), or you can wait some time for Marlin to revert to the status screen.

This and other problems with the knob suggests to use the knob only in Marlin mode, and use long touch to switch modes both from Marlin to TFT and from TFT to Marlin. Using the knob in the touch mode operates the Marlin menu or settings underground and can lead to unpredictable results.

VijitCoder, 2020-06-08:
> Using the knob in the touch mode operates the Marlin menu or settings underground and can lead to unpredictable results.

There is another side on the coin: if you **not** using the knob in the Touch mode you will be compelled to tip the screen a lot of times for setting up the numbered values (temperature, fan speed). I mean, you can't just pull the finger to the screen and wait while the desired value will be reached. I need to hitting. And this is inconvenient.

Rotating the knob without the seeing is dangerous: in the printing mode you can change Z offset. The best way here is to wait while the menu will closed by timer and the screen will updated. But if you was set up the long timeout (as I my case) then it will be a looong waiting :)

Tygrys-1, 2020-06-08:
To be on the safe side you do better not use knob. ;-)
I fully agree, that we need some convinient way to increase/decrease values in touch mode. Or make the knob invisible for the Marlin in such cases.

dp250f, 2020-06-17:
What we need is support for Marlin's HOST_ACTION_COMMANDS. If TFT mode supported that, things like M600 would work as expected. I submitted issue #736 for this.

ETE-Design, 2020-08-06:
@dp250f Look here... Just find a way to implement it :-) M600 Seems to be possible... https://www.youtube.com/watch?v=HJpfO8XifqI

bigtreetech, 2020-09-01:
Hello HOST_ACTION_COMMANDS should be worked after #984

closed 2020-09-01 by bigtreetech