bigtreetech/BIGTREETECH-TouchScreenFirmware #759: [Q] What is Printing.sys on the SD card?

VijitCoder, 2020-06-04:
**Your question or details you need. Please describe.**

I used a SD card. I started printing, then powered off the printer. And I found a file "Printing.sys" on the card.

I suspect this is a feature "Print after power failure", and the screen firmware constantly updates that file during printing. Am I right?

**specify the device variant related to your question.**

Bigtreetech TFT35-E3 V3.0
Firmware 26.x

VijitCoder, 2020-06-04:
I digged in the code. It really is what I suspected - _"Power Loss Recovery"_ (found the right naming).

Now the million dollar question: how to switch it off? I don't need this advantage at the cost of accelerated SD card degradation. I believe that this choice should be allowed for the user.

bigtreetech, 2020-06-08:
Hello, you need only modify here. https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/1cf14b2742c9b528b967a1e50dff0a60dd530796/TFT/src/User/Menu/PowerFailed.c#L27
`if(infoFile.source == BOARD_SD)  return false; // on board SD not support now`
to
`return false`
will disable this function.

We will add the switch of this function in the setting interface later.
Thank you for your suggestion

VijitCoder, 2020-06-08:
Roger that. Thanks.

bigtreetech, 2020-07-28:
Hello, There is a switch button in Feature Setting menu after #876, so close here

closed 2020-07-28 by bigtreetech