bigtreetech/BIGTREETECH-TouchScreenFirmware #760: [Q] Where should I connect the TFT28's Filament sensor?

poumsoon, 2020-06-04:
Where should I connect the TFT28's Filament sensor?
I couldn't find an answer anywhere.

Tygrys-1, 2020-06-05:
Does TFT28 support this?

oldman4U, 2020-06-05:
Hi poumsoon.

I have the same TFT but never thought it would support a filament sensor. What makes you believe it does so? Please let us know.

Thank you

poumsoon, 2020-06-08:
When using tft28, if the filament sensor is connected to the board, it does not work. Then, of course, I thought I should apply for tft lcd. As supported by tft35. If not, bigtreetec is making and selling ridiculous products.

oldman4U, 2020-06-08:
Hi.

There are boards and there are TFTs which support different kinds of
functions. But not all of them support everything , so the functions and
the price is different and the customer buys what fits his needs and budget
best.

I am sorry, but the TFT28 does not support for a filament sensor.

Please be so kind and close this ticket in case you do not need it any
more.

Thank you

poumsoon, 2020-06-08:
Then, if you use tft28, can you not use the filament sensor?
Then, it seems to be correct to move towards the bug or error.

Tygrys-1, 2020-06-08:
> When using tft28, if the filament sensor is connected to the board, it does not work.

What board, how it is connected, what does it mean "not working"?

As a rule - marketing say what the product **does** support, and not list all things that it **doesn't** do.

oldman4U, 2020-06-08:
Hello poumsoon.

It would be important to know, if the TFT28 has been advertised somewhere to support the filament runout sensor - this is why i asked you what makes you believe that this is the case.

Fact is, that it does not support it.

BTT could describe better what their products support and what not and in some cases they do already so. The SKR v1.3 to v1.4 charts are very good in this, but i agree with Tygrys-1, it is not the job of the advertisement agency to write what a product can not do.

I am afraid you bought the wrong product in case you would like to use a filament runout sensor with the TFT, but you would need one which supports this kind of functionality. 

It would be a bug, in case the function itself would be available, but does not work because of a hard- or software problem. But this is not the case.

poumsoon, 2020-06-08:
I am not English speaking, so I didn't understand exactly what you said.
I am a maker of many 3D printers. This time, BTT products are applied to produce 3D printers.
Currently, skr1.3 board is used and BTT TFT28 LCD is applied.
Filament sensor is an important function, and I am trying to apply it, but there is no way.
Why do you waste time and money making these products?

Tygrys-1, 2020-06-08:
Why not connect the sensor to SKR board?

oldman4U, 2020-06-08:
Hi poumsoon.

I am not a native English speaker too and we all are not working for BTT but try to help you.

Seems the SKR supports filament break, so why not connecting it there?

oldman4U, 2020-06-08:
I have to add, that I do not own this SKR board, but at least it is advertised to support it, so I guess this could work. The TFT28 I also own does not support it - for sure and I am still not sure what makes you believe that it does.

Tygrys-1, 2020-06-08:
I suspect, that poumsoon might be right here. But the problem might be not with the filament sensor not working on SKR board, but the inability of TFT screen to react to printer's message of the filament end. This information is displayed on Marlin's screen only, and the TFT (or other serial terminal) is not informed of it. See the #758

oldman4U, 2020-06-08:
Aaaaah, Tiger, i believe you are right and now this all makes sense ( i believe at least;-))

So this means that if you have a RS232 only TFT (without Marlin emulation mode) some features of the mainboard can not be used - right?

Tygrys-1, 2020-06-09:
@oldman4U You are right.
@poumsoon - I see a chance to connect the filament sensor to the TFT28, as it exposes some pins, but it will require TFT software modification, and in fact it wil not solve some other problems the printer user might encounter with TFT28.

poumsoon, 2020-06-09:
Hmm... I thought you'd give me an answer, so I'm sure you know this. The filament sensor works well on the 12864 LCD.
However, there is no way when using it with tft28.
After all, if tft28 is used, the filament sensor cannot be used at this time. If possible, support serial monitoring, but if that's difficult, it's best to use at least the extra port on the TFT28.

oldman4U, 2020-06-09:
Hi poumsoon.

It took some time for us to understand your question and until then we did not know if we would be able to help or not. But now we understand and the answer is, that using the current version of Marlin, the filament sensor can not be used together with the TFT28, sorry.

My personal opinion. I have both the TFT28 and TFT35 E3 v3 and i am about to replace the TFT28 with another 35. Why? Not that i need a filament sensor or that i want to control the power or that i would use any other of the fancy functions the 35 offers. It is also not the size of the screen, it is the speed, the ability to dimm the screen, the ability to use an USB stick (and WIFI) for printing and the buzzer.

Tygrys-1, 2020-06-09:
The problem is in fact not with the filament sensor working or not - it actualy is working, but Marlin is informing about that on it's own display, and not in terminal. So more or less the problem (or feature) of the Marlin.
I'm not BTT, I'm just a printer user and I have some fun with making things work better. As I'm not a STM32 guru, so no direct solution for you. On TFT28 I found exposed pins PA13 and PA14 on JTAG socket, so what I think should be done is to reconfigure one of them for direct I/O, enable the sensor support in firmware for TFT28 and set that pin as a sensor input. And there is a need for 5 pin plug for JTAG socket, and test if everything works. Ofcourse this will require printing from the SD from TFT, not from the mainboard. 
But - it will not solve any other interaction problem that Marlin handles through it's own display, like the filament change and maybe others.

poumsoon, 2020-06-09:
> Hi poumsoon.
>
> It took some time for us to understand your question and until then we did not know if we would be able to help or not. But now we understand and the answer is, that using the current [...]

HI, oldman4u.
I cannot understand exactly what you are saying.
What are you talking about?
I cannot understand exactly what you are saying.
What are you talking about?
In conclusion, do you think it is right to sell the product without warning that the filament sensor cannot be used when using TFT28 sold by BTT?

oldman4U, 2020-06-09:
Hi poumsoon.

I am sorry that you do not understand what i say. Like Tygrys-1, i am only a user who tries to help. Sometimes this works, sometimes it does not.;-(

Because the TFT28 does not support what you want to do, i recommended to you to try the TFT35 instead and listed a few additional benefits I can see, when i compare the two displays.

I believe it is right to advertise what a product can do and not what it can not do, except for comparison.

But i could imagine that it would be great to have a chart, comparing the functions of the BTT TFT displays which are available today. Maybe you could make this chart which would help others to avoid buying the wrong product. This would be great!

Tygrys-1, 2020-06-09:
I think that BTT should do a wiki about versions, connections, features and restrictions. In fact, this information is to be found on github in different places, but it is far from ease of find and use.
I was hoping for more magic from the touch mode, found more limitations and this is frustrating. So I understand other people in that matter.

Due to the hardware limitations the solution is not on the BTT side. I think, that it requires some extension on the Marlin side.

oldman4U, 2020-06-09:
Tygrys-1 i totally agree. Good documentation is missing and sometimes things which do not work together are sold as a "bundle". Happened to me with the SKR Pro and A4988 stepper drivers, which finally ended in an advanced documentation from GadgetAngel: https://github.com/GadgetAngel/SKR-PRO-V1.1-Stepper-Driver-Jumper-Configuration-Manual

 On the other hand, this all is Open Source, with all its advantages and disadvantages and there are closed systems available where you get a clearly defined solution out of the box . I prefer the open solution over the closed system, being aware that i have to invest more time on my side to make it work. 

The question is who is going to make the better documentation? A simple spreadsheet would be  a good starting point i believe.

Tygrys-1, 2020-06-09:
I think that the documentation could be done by someone, who has the adequate knoledge or experience to do it. I have just one TFT35 and I am slowly learning things here.
And the most competent should be someone from BTT. There is even not the problem of writting it, as all is here in repositories, it is the matter of organising this into easy accesible and readable form.

oldman4U, 2020-06-09:
Tygrys-1, like always - i agree. But you also know that sometimes those
learned all the things the hard way can explain things much better than the
experts. We could for example start with the framework by defining the
general format (structure) and fill out the things we know and leaving the
things we do not know undefined. I am sure we would then find other users
willing to add more information, depending on the hardware they use.

Check this out:
https://docs.google.com/spreadsheets/d/1BaNlA7yqBglrOInbjI6yAhvUzGHaxhHP0k_OGtMAZcQ/edit?usp=sharing

The question is how many users would read this before they buy something
and where to place it, so it is easy to "find" it.

poumsoon, 2020-06-10:
okay.
Tygrys-1 and OLDMAN4, thanks a lot for your answers.
It helped me to understand BTT products.
In conclusion, I will tell you my thoughts.
  The TFT28, which cannot use the filament runout sensing function, which is a very basic function for 3D printing, should not be sold, and if it is sold, a warning message stating that the filament sensing function cannot be used should be written.
  Some communities say that BTT TFT LCD is still a premature product. I agree with some, but if it only supports the filament sensing function, I think it is a product that is not lacking for general users to use.

closed 2020-06-10 by poumsoon