bigtreetech/BIGTREETECH-TouchScreenFirmware #762: Any way to re arrange this menu layout

MontyFlange, 2020-06-05:
It seems out of line to me :)

I would like it 

X  Y  Z
X  Y  Z

Not jumbled like it is ?

Thanks again

[image]

Tygrys-1, 2020-06-05:
I think good idea. Requires only to exchange Z- with X+.
Current layout is meant to be more "natural" to the real directions, but it is harder to memorize.

Switching 4 different lengths will be more anoying than 3, so I do not like the idea.
I'd rather like the option to move X and Y  to center/limit on long press - or use the "key repeat" on long press. 

For faster moves I use the knob, but beware side effects.

radek8, 2020-06-05:
A long press is to move to marlin mode.

Just swap this line to swap buttons.

[image]

Tygrys-1, 2020-06-05:
Long press to move to Marlin (from anywhere) is a waste of the interface potential funcionality. What is the reason to go to the Marlin mode from a movement sub menu?

Unless I'm missing something, it should be left for the knob long press only, or the long press on the main menu. But if we can "free" the long press from Marlin access, I see some additional potential we can have in the menus. I.e. long press on "back" can move to top menu, not only the level up.

radek8, 2020-06-05:
And how would you like to get from Marlin back to the touch screen?

Tygrys-1, 2020-06-05:
Like today - by long pressing the knob. And I see no reason to remove the todays ability to do this by long pressing the screen, as in Marlin mode the touch screen is not used.
Can the long press interpretation be context - sensitive? When called from screens where we use +/- buttons or back buttons or whatever we agree to use it for it will do some things, on main menu or in Marlin mode it will invoke the mode change screen.

Using the knob in touch mode to change values is a good idea, but has side effects on Marlin, so maybe we can eat a cake and still have a cake with this "long press on certain buttons" idea?

radek8, 2020-06-05:
And how do you get from the Merlin screen to a display that doesn't have a coder?
For example, BIGTREETECH TFT28
All variants have the same FW

http://www.bigtree-tech.com/shop/product/show/id/29

You alone higher that the use of a coder in touch mode has side effects.

radek8, 2020-06-05:
The idea is nice but there are HW limitations

MontyFlange, 2020-06-05:
I don't have a knob. So thats out

MontyFlange, 2020-06-05:
the previous version had them in the right order
[image]

Tygrys-1, 2020-06-05:
> 
> 
> I don't have a knob. So thats out

This is what I suspected :-)

Will it be any problem to you to go to Marlin mode from the main menu only, not just from anywhere as it is today?

Tygrys-1, 2020-06-05:
> 
> 
> The idea is nice but there are HW limitations

Sure they are.
Please explain.

MontyFlange, 2020-06-05:
> > I don't have a knob. So thats out
> 
> This is what I suspected :-)
> 
> Will it be any problem to you to go to Marlin mode from the main menu only, not just from anywhere as it is today?

I want to rearrange them, Why would going into Marlin mode help ?

Tygrys-1, 2020-06-05:
You have got an answer from Radek on rearangement.

I just asked about the need to use long press to go to the marlin mode from every menu, not just the main one. This is connected to your second question about speeding up the moves.

MontyFlange, 2020-06-05:
> You have got an answer from Radek on rearangement.
> 
Where ? I don't see it, just mentions about going to Marlin mode

Tygrys-1, 2020-06-05:
Radek's first post in this thread.
You have to modify the configuration.h file, comment the #define ALTERNATE_MOVE_MENU, rebuild the firmware and update it to TFT.

MontyFlange, 2020-06-05:
Ahh OK I don't know how to compile this firmware

radek8, 2020-06-05:
what is the display?

MontyFlange, 2020-06-05:
TFT35 V2

[image]

MontyFlange, 2020-06-05:
They are in the 'right' order with Unifi menu

radek8, 2020-06-05:
[BIQU_TFT35_APP1_V2.0.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4737562/BIQU_TFT35_APP1_V2.0.26.x.zip)

radek8, 2020-06-05:
Use Config.ini for configuration
Edit the config.ini file according to your own configuration, upload to the SD card, insert it into the display and restart.

MontyFlange, 2020-06-05:
Cool with give it a try in a bit . Thanks

oldman4U, 2020-09-14:
@MontyFlange 

Please close this ticket in case you do not need it anymore. 

Thank you

closed 2020-09-15 by MontyFlange