bigtreetech/BIGTREETECH-TouchScreenFirmware #763: [FR] Recover possibility to have the custom menu with icons.

GenaUser, 2020-06-05:
For frequent use, the custom menu by list is not practical.
I would like to recover the possibility of having a custom menu with icons, and to keep the Gcode menu by list.
Better, I am in favor of just adding a second customizable menu by icons without changing anything from the current version.
@bigtreetech is it possible?
Thanks in advance to look at this possibility

radek8, 2020-06-05:
use the config.ini file

[image]

radek8, 2020-06-05:
Edit the config.ini file according to your own configuration, upload to the SD card, insert it into the display and restart.

oldman4U, 2020-06-15:
Hello.

What is the status of the reported issue?

GenaUser, 2020-06-15:
Hello @oldman4U,  nothing has changed.
Status is open.

oldman4U, 2020-06-15:
Hi.

I spoke with an engineer and he confirmed that the ability to have Icons has been remove for the following reasons.

1. The number of Icons has been limited to 6 due to the number of pixels available on screen
2. There was only one generic Icon, which made this function useless until the user generated his own Icons.

Please let me know what you think

Thank you

GenaUser, 2020-06-15:
Why not allow the 2 possibilities (list and icons) according to our needs?
For example, my printer needs several specific functions that are constantly used.
They cannot be used in good conditions in list mode, such as switching head, stop motors , move to the nozzle change position ...

[image]

oldman4U, 2020-06-15:
This is most likely because there is currently no way to have several pages of the same menu, so buttons will only work if the number of icons exceed the space which is available. In list mode, there is this capability by using page up/down a similar function is needed for Icon view and it would be great to get this function available with a new version.

GenaUser, 2020-06-15:
Ideally, have an icon to access the personalized menu by list, and another to access the personalized menu by icons.
May also have an option to hide or show these 2 menus.

oldman4U, 2020-06-15:
Please keep in mind that space is limited and optional options are luxury. The same time this makes things more complicated than needed, so the best way to achieve the solution is the way to go.

Functionality wise the list mode does the job, but there are situations where buttons are better. Maybe you could make some sketches for the developers showing the best solution you could imagine which works best for all users.

GenaUser, 2020-06-15:
Currently, the customizable menu is displayed in two places; in "menu" and in "machine".
I suggest using the customizable menu by icons in "menu" (frequently used functions) and the customizable menu by list in "machine" for the other functions.

I'm not looking for new features, but only to find the one that has been removed. Features that determined the choice of this screen.

[image]

oldman4U, 2020-06-22:
Hi.

Sorry for the late answer.

The two Custom under Machine and Menu are showing the same functions and therefore they are redundant. I would not count on it that such redundant functions will stay forever. So there has to be a solution which allows to have more than 6 (or seven in case there is only the Back button in addition) buttons in a single interface. Do you have an idea, how button based custom gcode interfaces could look like, where the maximum of 15 commands is used?

GenaUser, 2020-06-22:
Hi,
My idea is to use the same interface with 7 icons in the first custom menu, and with the 15 commands list in the second.

Configuration.h might look like the following:
/**
* Custom G-code Commands by icons (Menu/Custom1)
* Support up to 7 custom gcodes, uncomment CUSTOM_X_LABEL and CUSTOM_X_GCODE to enable custom gcode
 * CUSTOM_X_LABEL is the name of the custom button, CUSTOM_X_GCODE
 * CUSTOM_X_GCODE is the gcode to be sent by the custom button, end with '\n'
 * You also need to customize the icon corresponding to the command
 * Copy your custom icon to the SD card to be updated, such as:"TFT35/bmp/Custom0.bmp", "TFT24/bmp/Custom1.bmp", etc...
 * The format of the custom icon is as follows
 * Bit depth: 24 / 32 bit, Pixel size: 95*95(for TFT35), 70*70(for TFT28/TFT24)
 */
#define CUSTOM_0_LABEL "Z => 0"
#define CUSTOM_0_GCODE "G1 Z0\n"
#define CUSTOM_1_LABEL "Buse G"
#define CUSTOM_1_GCODE "T0\n"
#define CUSTOM_2_LABEL "Buse D"
#define CUSTOM_2_GCODE "T1\n"
#define CUSTOM_3_LABEL "AR Moteurs"
#define CUSTOM_3_GCODE "M18\n"
#define CUSTOM_4_LABEL "Z => 10"
#define CUSTOM_4_GCODE "G1 Z10\n"
#define CUSTOM_5_LABEL "Origines"
#define CUSTOM_5_GCODE "G28\n"
#define CUSTOM_6_LABEL "Changer"
#define CUSTOM_6_GCODE "G28\n G1 X165 Y0\n"

/**
 * Custom G-code Commands by list (Menu/Settins/Machine/Custom2) *
 * Support for up to 15 custom G-codes. Uncomment CUSTOM2_*_LABEL and CUSTOM2_*_GCODE to enable custom G-code.
 *
 * CUSTOM2_X_LABEL is the name of the custom button, CUSTOM2_X_GCODE is the G-code to be sent by the custom button, this should always end with a New-Line character '\n'
 */
#define CUSTOM2_0_LABEL "Disable Steppers"
#define CUSTOM2_0_GCODE "M84\n"
#define CUSTOM2_1_LABEL "Init SD Card"
#define CUSTOM2_1_GCODE "M21\n"
#define CUSTOM2_2_LABEL "Release SD Card"
#define CUSTOM2_2_GCODE "M22\n"
#define CUSTOM_3_LABEL "Enable Leveling State"
#define CUSTOM2_3_GCODE "M420 S1\n"
#define CUSTOM2_4_LABEL "Save to EEPROM"
#define CUSTOM2_4_GCODE "M500\n"
#define CUSTOM2_5_LABEL "Restore from EEPROM"
#define CUSTOM2_5_GCODE "M501\n"
#define CUSTOM2_6_LABEL "EEPROM Defaults"
#define CUSTOM2_6_GCODE "M502\n"
//#define CUSTOM2_7_LABEL "Custom7"
//#define CUSTOM2_7_GCODE "M105\n"
//#define CUSTOM2_8_LABEL "Custom8"
//#define CUSTOM2_8_GCODE "M105\n"
//#define CUSTOM2_9_LABEL "Custom9"
//#define CUSTOM2_9_GCODE "M105\n"
//#define CUSTOM2_10_LABEL "Custom10"
//#define CUSTOM2_10_GCODE "M105\n"
//#define CUSTOM2_11_LABEL "Custom11"
//#define CUSTOM2_11_GCODE "M105\n"
//#define CUSTOM_12_LABEL "Custom12"
//#define CUSTOM_12_GCODE "M105\n"
//#define CUSTOM_13_LABEL "Custom13"
//#define CUSTOM_13_GCODE "M105\n"
//#define CUSTOM_14_LABEL "Custom14"
//#define CUSTOM_14_GCODE "M105\n"

It is also possible to use a single menu with several pages. 6 customizable icons with a back and next page icon.

oldman4U, 2020-06-23:
OK, got it.

oldman4U, 2020-10-20:
Hello GenaUser.

Could you please be so kind and close this ticket. Your feature request is covered in the pinned Feature Request ticket #1177 

Thank you

closed 2020-10-20 by GenaUser