bigtreetech/BIGTREETECH-TouchScreenFirmware #766: [BUG] Can not remove "LCD12864 Simulator" text on the top on Marlin mode

DJDOUBLEG, 2020-06-06:
I try this settings in Configuration.h
/**
 *  Text displayed at the top of the TFT in Marlin Mode.
 */
#define ST7920_BANNER_TEXT "Some text"

/**
 *  show banner text at the top of the TFT in Marlin Mode.
 */
#define ST7920_SHOW_BANNER false //to enabled: true | to disabled: false

And recompie reflash, then no change.

trevjonez, 2020-06-06:
I had to "reset" from the touch menu to get some other changes I made to configuration to be applied (fg/bg colors etc...) 

It would seem the onboard storage wins over changes to the defaults in firmware. Most likely to ensure a firmware update doesn't destroy any customizations by the user.

Tygrys-1, 2020-06-07:
Firmware defaults are now overwriten by the config.ini setings.
Try to find settings for this in config.ini, edit, copy to sd and reboot the printer.

DJDOUBLEG, 2020-06-07:
After recompile/reflash make setting reset, then text is removed.
Thank you!

closed 2020-06-07 by DJDOUBLEG

AntoszHUN, 2020-06-16:
This is not a solution to stupidity. Why can't it work the way it used to? Why put everything in the config-ini file? What is the point of this? What is the meaning of configuration.h then?

DJDOUBLEG, 2020-06-16:
Mint a smoothieware vagy a klipper. Ujdonsag. Kellett a 43 50 es a 70 tft
konfiguralasahoz. Ne kelljen minden alkalommal uj forditani.

Zs.Antal <notifications@github.com> ezt írta (időpont: 2020. jún. 16., K
19:24):