bigtreetech/BIGTREETECH-TouchScreenFirmware #767: [Q] Disable "LCD12864 Simulator"

iMattmax, 2020-06-07:
I  use the latest firmware for the TFT35 and wanna know how I can disable the overlay "LCD12864 Simulator" on the Marlin screen. 
As I get this TFT35 I think there was no overlay, but after the update there is now an overlay and I wanna get rid of it.

carl1961, 2020-06-07:
in config.ini around line 110  you get the settings, you can turn it off or rename it. you put this on your sdcard and reboot TTF  
#### Marlin Mode show title
# Options: [enable: 1, disable: 0]
marlin_show_title:1

# Run Marlin Mode in Fullscreen
# Options: [enable: 1, disable: 0] [Disabled. RECOMMENDED FOR TFT24]
marlin_fullscreen:0

#### Text displayed at the top of the TFT in Marlin Mode.
marlin_title:LCD12864 Simulator

iMattmax, 2020-06-07:
Do I flash than only the ini or the whole firmware again?

radek8, 2020-06-07:
Edit the config.ini file according to your own configuration, upload to the SD card, insert it into the display and restart.

iMattmax, 2020-06-07:
Thanks, I will try it later

oldman4U, 2020-06-08:
Hi. 

Please close unneeded tickets. 

Thank you

closed 2020-06-08 by iMattmax