bigtreetech/BIGTREETECH-TouchScreenFirmware #769: Create Button for Filament loading

joydi, 2020-06-07:
Hi,

i have an question did somebody mange to create an button to load filament and onload it ?

When yes can you provide me how this works ?

Thanky in advance

oldman4U, 2020-06-08:
Hi joydi.

There is a filament load and unload button which can be used as soon as the nozzle is heated up. But i guess you have a bowden setup and you would like to use it to change the filament, right?

Maybe this can be done using custom code commands.

joydi, 2020-06-08:
Hey Oldman4u,

yes indeed i have an bowden setup and like to say hey load me please 400m filament or unload it.
Is it possible ?

oldman4U, 2020-06-08:
Have you tried to use custom gcode commands to make this happen.? Guess, you would have to heat the nozzle first and then do the extrusion. What kind of display do you have? I used this function for some time using the Marlin mode, where this can be done.

Tygrys-1, 2020-06-08:
You may use a "Custom gcode" for this. I suggest not to use M701/M702, at it requires interaction with the Marlin's display, and it is not accesible in touch mode. But you may try it. 
Rather use the sequence like:
M104 S[your temperature] ;set depending on filament kind
M109 S[5 degree lower than above] ;wait for heatup
M83 ;Extruder relative
G1 E[a little] ;on retraction move forward a little to soften the tip of the filament and count from the nozzle.
G1 E[your length] F[your speed] ; actual retraction/feed 

Build your sequence, write into config.ini at empty slot, name whatever you like (i.e. "Load ABS"), copy the file to sd, put into the TFT and turn on the printer.

custom_label_8:Unload PLA
custom_gcode_8:M104 S200\nM109 S195\nM83\nG1 E10 F600\nG1 E-470 F600\n

The drawback is, that you will not see any feedback from the TFT, just temperature rising and then retract or feed on the extruder. You may also add Z rise a little and lock the motors, to stabilize the head during extruder operation.

oldman4U, 2020-06-09:
Hello Tiger, joydi.

@joydi Do you know how to add custom gcode to your firmware?

zalexzperez, 2020-07-24:
This is not precisely related to filament change, but I hope I can help someone out there.

I installed the BTT TFT35 E3 V3.0 today on my Ender 3 and quickly realized there were no filament change actions nor did the printer let me continue printing after an M600. My finding concerns the latter.

By mistake, I realized that when the printer encounters an M600 and it's done loading the new filament, it gets stuck at the PURGE MORE (default selection) and CONTINUE dialog which is not user visible in the touchscreen mode..but you can actually clockwise rotate the encoder and press it as if you had a standard reprap LCD, and there you go, the printing continues.

oldman4U, 2020-07-25:
@zalexzperez Thank you for your help

@joydi Please close this ticket or answer our replies. Thank you

closed 2020-09-01 by bigtreetech